t-oster/VisiCut #506: debian .deb package incompatibility issues

eMPee584, 2019-05-24:
Packages work with `dpkg -i` and `dpkg-deb -I`, but `ar` and `gdebi` choke on it..  will post logs later.

eMPee584, 2019-05-25:
```
ar x visicut*.deb =>
ar: visicut_1.8-64-gc37034f0-1_all.deb: malformed archive
ar: visicut_1.8-69-gf5421647-1_all.deb is not a valid archive
```
from `live-build` log:
```
E: Invalid archive member header […]
E: Errors apply to file './visicut_1.8-64-gc37034f0-1_all.deb'
E: Archive is too short
E: Errors apply to file './visicut_1.8-69-gf5421647-1_all.deb'
```

mgmax, 2019-05-25:
This is strange. We just use `checkinstall`, nothing special:
https://github.com/t-oster/VisiCut/blob/f5421647c77c2f91150f0908034057dba6c9668a/distribute/distribute.sh#L118
Maybe it's a bug in checkinstall?

BTW, if you are building something like a live image with VisiCut etc., I'd also be interested in that.

eMPee584, 2019-05-25:
well i'm trying :rofl: 
 .. repacking it worked so inclusion in packages.chroot did successfully install it without further change
the ar compression seems to be messed up in some subtle way :thinking:

eMPee584, 2019-06-29:
So as even the newest packages (1.8-77) still is 'malformed', I finally checked out this repo and ran `distribute.sh`. It produced a perfectly fine deb, so there might be some problem with your toolchain?
I'm on debian unstable, so binutils (which provides ar) is at 2.31.1-16 ..

t-oster, 2019-06-30:
Can you run the visicut-builder docker container? https://github.com/t-oster/VisicutBuilder

gerdg-dd, 2019-07-01:
Generate Packages and Packages.gz ( debian buster)

apt-ftparchive packages . > Packages

E: Invalid archive member header exception 'PDOException' with message 'SQLSTATE[HY000]: Gene
E: Errors apply to file './visicut_1.8-77-g840df7cc-1_all.deb'

eMPee584, 2019-07-29:
`ar: visicut_1.8-80-g681e741c-1_all.deb: malformed archive` …
does the builder container produce the debs for https://download.visicut.org/?
which binutils do you use? more likely that is the source of the problem, checkinstall is just a giant bash script..

t-oster, 2019-07-29:
yes I am using that very container to build the downloadable images

eMPee584, 2019-07-29:
Ok with the updated container, it builds successfully and the resultant `visicut_1.8-80-g681e741c-1_all.deb` is valid!

The broken file from https://download.visicut.org/ has got trash at the end:
```
exception 'PDOException' with message 'SQLSTATE[HY000]: General error: 14 unable to open database file' in /var/www/html/bootstrap.php:35
Stack trace:
#0 /var/www/html/bootstrap.php(35): PDO->exec('INSERT INTO dow...')
#1 /var/www/html/download.php(43): update_db('branch-master/D...')
#2 {main}
```
Something is b0rked on the build machine?

t-oster, 2019-07-29:
thank you for figuring that out. I will fix that tomorrow

t-oster, 2019-07-31:
Is fixed now. I had some wrong access rights for a parent folder, which prevented the sqlite driver from opening the database file.

closed 2019-07-31 by t-oster