t-oster/VisiCut #512: K40 Integration, misses dependencies.

tatarize, 2019-06-19:
I deleted everything on the laser machine and tried the nightly. 

```
Exception in thread "Thread-7" java.lang.NoClassDefFoundError: org/usb4java/LibUsbException
	at com.t_oster.liblasercut.drivers.K40NanoDriver$K40Queue.open(K40NanoDriver.java:1319)
	at com.t_oster.liblasercut.drivers.K40NanoDriver$K40Device.open(K40NanoDriver.java:464)
	at com.t_oster.liblasercut.drivers.K40NanoDriver$K40Device.open(K40NanoDriver.java:458)
	at com.t_oster.liblasercut.drivers.K40NanoDriver.sendJob(K40NanoDriver.java:95)
	at com.t_oster.visicut.VisicutModel.sendJob(VisicutModel.java:786)
	at com.t_oster.visicut.gui.MainView$65.run(MainView.java:2275)
Caused by: java.lang.ClassNotFoundException: org.usb4java.LibUsbException
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 6 more

```

Some file didn't get updated.

tatarize, 2019-06-19:
https://github.com/t-oster/VisiCut/blob/master/nbproject/project.properties

Needs to also note include the new usb4java dependencies, even though they are dependencies of dependencies.

closed 2019-06-19 by t-oster

t-oster, 2019-06-19:
yes we should have switched to maven a long time ago, but some dependencies are missing and I just did not get them into maven central with my limited time.

t-oster, 2019-06-19:
could you verify that visicut_1.8-75-g7a4c73e4 works?

tatarize, 2019-06-19:
@t-oster I can confirm that it does not.

Exception in thread "Thread-5" java.lang.NoSuchMethodError: java.nio.IntBuffer.clear()Ljava/nio/IntBuffer;
	at com.t_oster.liblasercut.drivers.K40NanoDriver$K40Usb.update_status(K40NanoDriver.java:1539)
	at com.t_oster.liblasercut.drivers.K40NanoDriver$K40Usb.wait(K40NanoDriver.java:1596)
	at com.t_oster.liblasercut.drivers.K40NanoDriver$K40Usb.wait_for_ok(K40NanoDriver.java:1589)
	at com.t_oster.liblasercut.drivers.K40NanoDriver$K40Queue.execute(K40NanoDriver.java:1376)
	at com.t_oster.liblasercut.drivers.K40NanoDriver$K40Device.execute(K40NanoDriver.java:767)
	at com.t_oster.liblasercut.drivers.K40NanoDriver.sendJob(K40NanoDriver.java:276)
	at com.t_oster.visicut.VisicutModel.sendJob(VisicutModel.java:786)
	at com.t_oster.visicut.gui.MainView$65.run(MainView.java:2275)

The Usb4java library requires the data be packed into databuffers. These are absent.

tatarize, 2019-06-19:
Not absent, weirdly different between versions of java. 

https://jira.mongodb.org/browse/JAVA-2559

`The solution is to cast ByteBuffer instances to Buffer before calling the method.` It's something with the versioning and the version of java and change in the methods there. I was sure to use the version the manifest wanted for LibLaserCut even to the point of not being allowed the project COIN fixes for things like explicit diamond bracket classes.

t-oster, 2019-06-19:
Now?

tatarize, 2019-06-19:
It works!

Tossed everything in the trash.
Downloaded latest version.
Ran the jar file.
Installed Example settings.
Added K40.
Selected K40.
Imported a picture of a cactus.
Sized and positioned it.
Set "Engrave Everything"
Hit Execute.
Laser started firing away.
I now have a rastered image of a cactus.

tatarize, 2019-06-19:
There's a bunch of stuff whisperer needs to do to make sure libusb works on a system. This setup has those files nicely wrapped up and included into the jars. But, there might be something there that won't work out of the box for people. But, this is enough to send a clarion call for some beta testers.

tatarize, 2019-06-19:
I am running a copy of Raspberry Pi on this system, that was already setup running Whisperer previously with all the nonsense needed to get those drivers working predone. I'm not sure if the jars will handle that or some extra setup info will be needed. But, it certainly can work out of box. Which is what I needed it to.

t-oster, 2019-06-19:
Thank you for all that testing.

tatarize, 2019-06-19:
Needed that testing it so that I could ask folks elsewhere for testing. Need to see if there's any outstanding problems I'm missing before I take my expertise elsewhere. I'd like to know other people have few to no issues that require deep understanding. As those would be, currently, unfixable without me.

commit cf0e0c3d (2019-06-19): update nbproject. fixes #512
commit 7a4c73e4 (2019-06-19): update nbproject. fixes #512