t-oster/VisiCut #515: The use of xstreams for most settings causes fragility

tatarize, 2019-07-29:
XStreams works okay when you want settings just done, but the serialization of the values makes it so that there's little wiggle room in changing anything. For example the drivers are themselves laser properties, and they get their data saved with an XStream. But, if they wanted to shift over to a AbstractLaserProperty class it would cause the setting to crash and fail to load.

For the raster settings this even includes the specific class:
`class="com.t_oster.liblasercut.dithering.FloydSteinberg"` which effectively makes any attempts to alter those methods force it to break everybody's settings.  There's no methods for authors to write their own versions of these or correct anything during a transition since it's serialized from the state of the classes themselves any change to the class will break the settings. 

This is especially weird with the drivers since they are required to provide a list of their settings and the names of these settings etc. Seamless transitions to a different methodology will still break it, even if those functions given returned the exact same things.

This wouldn't be too bad except that when the settings break the element is just gone. *poof* rather than forcably reset to default. You change the way it does the dithering and it breaks the engrave.xml which makes that setting no longer exist as a thing anymore.

t-oster, 2019-11-01:
I agree. We should hook somehow into the deserialization method and allow for either altering the input before it's fed into xstream or the output before it's given back, so we can add config upgrades. Maybe we should also store the visicut version number that created the settings, so we can have versioned migrations?

tatarize, 2019-11-01:
Even if you don't do anything else, add the version numbers. Always add version numbers to things even if you rarely use them or never hook them up to anything. If you want that in the future, you'll wish you had done so in the past.

The solution it has currently is fast and easy, but it is not very tolerant, and tied my hands even when the data I was saving was by definition giving you the same information as before, the method the information was stored in was a problem. I think you'd need to actually read the properties the drivers give you, and only them, rather than serializing the data. And it might be nice if one of the properties it gave you back was the version number. Then if the drivers do something else they can migrate from previous settings.

t-oster, 2019-11-04:
I think this https://wiki.pmease.com/display/xmt/Getting+Started looks very promising

t-oster, 2019-11-05:
I successfully integrated the xmt library in a local branch of mine. It adds a version to all xml elements representing a class and the classes can have migrateXXX methods which can alter the xml before deserializing. I think this will fix most problems. However since my changes depend on #537, you will have to wait a bit.

t-oster, 2019-11-05:
@tatarize I pushed the branch https://github.com/t-oster/VisiCut/tree/feature-xmt for you. Please try if you can compile and run this version (you need to mvn install the maven branch of LibLaserCut first).

Then you can experiment with your changes with regards to migrating old setting files. You can create methods like 
```java
public void migrate1(VersionedDocument dom, Stack<Integer> versions) {
    //add default value for new property
   dom.getRootElement().addElement("someProperty").setText("new default");
   //rename property
   dom.getRootElement().element("oldPropertyname").setName("newPropertyName");
}
```
NOTE: in contrast to the tutorial I want the methods to be public and migrate1 is called on all xml that does not contain a version number.

tatarize, 2019-11-05:
I'll check into whether I can easily migrate the K40 driver settings over to AbstractLaserProperty without it breaking if I can find where I put that.  But, I currently don't have my head around it enough to know if it'll fix things. The issue I ultimately had was I gave a different datastructure to read the exact same settings and broke the settings. 

While being able to migrate settings might work fairly well, I'd think it would need to be able to not look inside my class and realize I changed the underlying methodology. The AbstractLaserProperty class I added was generally predicated on the idea that you could seamlessly swap out how the settings were stored and just have it work. But, rather because it was stored in a different manner it broke. Seems like it either needed to not ask me to save settings (just save those with serialization) or use just the settings I gave it.

t-oster, 2019-11-12:
the changes are now in master and everything is migrated to maven. if you cannot implement your refactorings with the help of such methods, please reopen.

closed 2019-11-12 by t-oster

mgmax, 2019-11-17:
Maybe an easier option for LaserDevice and LaserProperty would be to just use the properties shown in the GUI for serialization?
 Then, serialization would just save the key-value-list of properties and values, and deserialization would try to call the default constructor and then apply every property found in the file.

Most cases for migration would be solved silently:
If a new property is added to a class, it is initialized with a default value.
If a property is removed, setProperty() would be ignored.

If we really need more complicated migration, setProperty() could be modified to redirect old property names to new ones. 

This would match the user expectation that the settings are equivalent to the property table shown in the GUI.

tatarize, 2019-11-18:
That's not really serialization. That's saving the properties as given by the drivers. Which is clearly going to be preferred. The problem is serialization looks into the class and checks how it's actually setup. The problem there is that if I want to change that it breaks the settings. When really if I give it the exact same data it should have no troubles with that.

t-oster, 2019-11-18:
I agree saving the properties is the correct solution. In fact the LaserProperties class was intended to be just a typesafe representation of that, kind of a data class. Still there should be a version property and an option for drivers to hook in migration methods.

t-oster, 2019-11-18:
Since we use xstream for all of the settings in visicut the Migrations are much more powerful. You could now theoretically switch out drivers during migrations, supporting old settings from the first oldest version upwards.