t-oster/VisiCut #518: XML deserialization fails on latest Ubuntu 18.04

stesie, 2019-08-07:
We've got Visicut installed with a Epilog ZING cutter w/ calibrated camera.  The setup worked fine until most recent security update on Ubuntu 18.04 to openjdk-11-jre version `11.0.4+11-1ubuntu2~18.04.3`

VisiCut now fails to start correctly due to a xstreams exception `Homography as it does not have a no-args constructor`

```
WARNUNG: Cannot construct com.t_oster.visicut.misc.Homography as it does not have a no-args constructor : Cannot construct com.t_oster.visicut.misc.Homography as it does not have a no-args constructor
---- Debugging information ----
message             : Cannot construct com.t_oster.visicut.misc.Homography as it does not have a no-args constructor
cause-exception     : com.thoughtworks.xstream.converters.reflection.ObjectAccessException
cause-message       : Cannot construct com.t_oster.visicut.misc.Homography as it does not have a no-args constructor
class               : com.t_oster.visicut.misc.Homography
required-type       : com.t_oster.visicut.misc.Homography
converter-type      : com.thoughtworks.xstream.converters.reflection.ReflectionConverter
path                : /laserDevice/cameraHomography
line number         : 14
class[1]            : com.t_oster.visicut.model.LaserDevice
version             : null
-------------------------------
Aug. 07, 2019 6:30:53 NACHM. com.t_oster.visicut.managers.FilebasedManager readObjectFromXmlStream
WARNUNG: Failed to load object from XML.
Error reading: /home/stesie/.visicut/devices/Epilog_32_ZING.xml. Invalid File Format (created with old VisiCut version?)
```

Unfortunately I don't know about xstream to debug any further.  Any help appreciated

t-oster, 2019-08-07:
Looks like we need do change the homography class. Should be easy...

t-oster, 2019-08-09:
Can you provide a zip of your .visicut folder?

t-oster, 2019-08-09:
Can you retry with the version from in 5 minutes?

stesie, 2019-08-09:
Sure.  I'll try it out, next time I'm in the lab ...

stesie, 2019-08-12:
Hej, so I've tried things out now.  Turns out if I try the latest release, aka 1.8-90, it does *not* work. However this one doesn't ship the new xstream version (as per #522). If I replace it with the new xstream version it indeed works.

So once the new xstream lib is actually shipped, this can happily be closed.

And regarding the .visicut folder, if you're still interested: https://drive.google.com/file/d/1HhwUyAKOe1KeRnLTued_6klGYYx1E786/view
(ignore the Redsail cutter config file which fails with the default liblasercut)

t-oster, 2019-08-13:
Sorry the build server had a problem with pulling. The latest release should work now.

closed 2019-08-13 by t-oster