t-oster/VisiCut #521: Windows 10 launcher doesn't work

Arrri, 2019-08-09:
I cannot run VisiCut.exe on Windows 10. I cannot get any logs either. When I try to run it nothing happens.

I am able to run VisiCut by executing .jar from platform independent zip.

Arrri, 2019-08-09:
Java was installed with the visicut. It is in the PATH:

>java -version

java version "1.8.0_66"
Java(TM) SE Runtime Environment (build 1.8.0_66-b18)
Java HotSpot(TM) Client VM (build 25.66-b18, mixed mode, sharing)

t-oster, 2019-08-10:
If you run visicut.exe from the cmd it does not output anything?

Arrri, 2019-08-10:
Yes, I tried different options and never got any logs or any output at all.

The command finishes immediately, there is a blank line and the prompt is
ready for the next command:

> VisiCut.exe

>

Arrri, 2019-08-11:
[image]

t-oster, 2019-08-13:
Hi, can you retry with the latest version. However I did not change anything. Which version did work last?

Arrri, 2019-08-13:
Sure, I will.

I don't know this exact version (I will check it after work). About week ago I tried most of the available version and none of them worked for me. I started using jar from portable directly and I didn't have any problem with it.

I wanted to try inkscape plugin and I need exe version for that. Exe didn't work from the same portable version from which jar worked.

Arrri, 2019-08-13:
Nothing changed.
[image]

I tried versions:
1.7-310
1.8-75
1.8-80
1.8-90
1.8-92

Installer is unable to modify PATH env var but I've always used the default path and I then I manually added the path to PATH var:
PATH=....;C:\Program Files (x86)\VisiCut;...

Kr4cker, 2019-09-22:
I got exactly the same error.
Another problem which is probably related to this is that the inkscape plugin is not working.
It is not possible to add or open a sketch to/in VisiCut.
I used also the newest version of 1.8-92.
The java version was installed during the VisiCut installation.

mgmax, 2019-10-02:
I'm sorry but I am unable to reproduce this issue. I set up Windows 10 Pro in a VirtualBox VM with German language, installed Visicut and it starts correctly. Can you please tell me how to reproduce the bug, preferably on a clean installation? (You don't need a Windows license key for testing in a VM, just use the ISO from Microsoft and say "I dont have a product key", this will give you a few days of demo license for testing).

For reference, the launcher starts VisiCut with `javaw.exe -Xms256m -Xmx1024m -classpath ".;lib" -jar Visicut.jar --singleinstanceport 6543`

mgmax, 2019-10-20:
Please test again with version 1.8-99, which includes its own Java runtime.

Arrri, 2019-10-21:
I'll try to test it soon and let you know about the results.
*Alan Kutniewski*

Arrri, 2019-10-21:
I tested 1.8-99 and it works.

closed 2019-10-21 by t-oster

Kr4cker, 2019-10-22:
Same here. Installed 1.8-99 direct over the older version.
Now it works with the windows launcher =)
Thank you very much