t-oster/VisiCut #524: Can not open svg with pixel graphics in Ubuntu 18.04

Kr4cker, 2019-09-15:
## Description of the problem
I can not open a _svg_ file which contains a picture **and** a square (a square with no filling just a red outline and a [cat photo](https://en.wikipedia.org/wiki/Cat_intelligence#/media/File:An_up-close_picture_of_a_curious_male_domestic_shorthair_tabby_cat.jpg) inside).
The _svg_ file is generated by Inkscape v0.92.4.
If I import **only pixel** graphic or **only vector** graphic visicut works like a charm.
I also installed the oracle java version but I got the same error.

If I switch to Windows 10 where I installed the same version of visicut, everything works like it should.

## Troubleshooting
If I start visicut with the terminal 
```
cd /usr/share/visicut
java -Xms256m -Xmx2048m -jar Visicut.jar
```
I get the following error messages:
```
Gtk-Message: 14:12:44.830: Failed to load module "pantheon-filechooser-module"
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.thoughtworks.xstream.core.util.Fields (file:/usr/share/visicut/lib/xstream-1.4.11.1.jar) to field java.util.TreeMap.comparator
WARNING: Please consider reporting this to the maintainers of com.thoughtworks.xstream.core.util.Fields
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Security framework of XStream not initialized, XStream is probably vulnerable.
Security framework of XStream not initialized, XStream is probably vulnerable.
Security framework of XStream not initialized, XStream is probably vulnerable.
Security framework of XStream not initialized, XStream is probably vulnerable.
Security framework of XStream not initialized, XStream is probably vulnerable.
```
And if I try to load my test svg file I get the final error messages
```
Sep 15, 2019 2:09:02 PM com.kitfox.svg.SVGUniverse loadSVG
WARNUNG: Could not load SVG svgSalamander:/test_95_mit_95_ebenen.svg
java.lang.NoClassDefFoundError: sun/misc/BASE64Decoder
	at com.kitfox.svg.app.data.Handler$Connection.<init>(Handler.java:72)
	at com.kitfox.svg.app.data.Handler.openConnection(Handler.java:108)
	at java.base/java.net.URL.openConnection(URL.java:1063)
	at java.base/java.net.URL.openStream(URL.java:1129)
	at java.desktop/javax.imageio.ImageIO.read(ImageIO.java:1405)
	at com.kitfox.svg.SVGUniverse.registerImage(SVGUniverse.java:264)
	at com.kitfox.svg.ImageSVG.build(ImageSVG.java:137)
	at com.kitfox.svg.ImageSVG.updateTime(ImageSVG.java:352)
	at com.kitfox.svg.Group.updateTime(Group.java:309)
	at com.kitfox.svg.Group.updateTime(Group.java:309)
	at com.kitfox.svg.SVGRoot.updateTime(SVGRoot.java:296)
	at com.kitfox.svg.SVGDiagram.updateTime(SVGDiagram.java:230)
	at com.kitfox.svg.SVGUniverse.loadSVG(SVGUniverse.java:593)
	at com.kitfox.svg.SVGUniverse.loadSVG(SVGUniverse.java:487)
	at com.kitfox.svg.SVGUniverse.loadSVG(SVGUniverse.java:478)
	at com.t_oster.visicut.model.graphicelements.svgsupport.SVGImporter.importSetFromFile(SVGImporter.java:132)
	at com.t_oster.visicut.model.graphicelements.svgsupport.SVGImporter.importSetFromFile(SVGImporter.java:370)
	at com.t_oster.visicut.model.graphicelements.AbstractImporter.importFile(AbstractImporter.java:37)
	at com.t_oster.visicut.model.graphicelements.GraphicFileImporter.importFile(GraphicFileImporter.java:88)
	at com.t_oster.visicut.VisicutModel.loadGraphicFile(VisicutModel.java:711)
	at com.t_oster.visicut.VisicutModel.loadFile(VisicutModel.java:490)
	at com.t_oster.visicut.gui.MainView.loadFileReal(MainView.java:1871)
	at com.t_oster.visicut.gui.MainView.access$5400(MainView.java:136)
	at com.t_oster.visicut.gui.MainView$61.run(MainView.java:1788)
Caused by: java.lang.ClassNotFoundException: sun.misc.BASE64Decoder
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:583)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
	... 24 more

com.t_oster.visicut.model.graphicelements.ImportException: com.t_oster.visicut.model.graphicelements.ImportException: java.lang.NullPointerException
	at com.t_oster.visicut.model.graphicelements.svgsupport.SVGImporter.importSetFromFile(SVGImporter.java:375)
	at com.t_oster.visicut.model.graphicelements.AbstractImporter.importFile(AbstractImporter.java:37)
	at com.t_oster.visicut.model.graphicelements.GraphicFileImporter.importFile(GraphicFileImporter.java:88)
	at com.t_oster.visicut.VisicutModel.loadGraphicFile(VisicutModel.java:711)
	at com.t_oster.visicut.VisicutModel.loadFile(VisicutModel.java:490)
	at com.t_oster.visicut.gui.MainView.loadFileReal(MainView.java:1871)
	at com.t_oster.visicut.gui.MainView.access$5400(MainView.java:136)
	at com.t_oster.visicut.gui.MainView$61.run(MainView.java:1788)
Caused by: com.t_oster.visicut.model.graphicelements.ImportException: java.lang.NullPointerException
	at com.t_oster.visicut.model.graphicelements.svgsupport.SVGImporter.importSetFromFile(SVGImporter.java:152)
	at com.t_oster.visicut.model.graphicelements.svgsupport.SVGImporter.importSetFromFile(SVGImporter.java:370)
	... 7 more
Caused by: java.lang.NullPointerException
	at com.t_oster.visicut.model.graphicelements.svgsupport.SVGImporter.importSetFromFile(SVGImporter.java:133)
	... 8 more
com.t_oster.visicut.model.graphicelements.ImportException: com.t_oster.visicut.model.graphicelements.ImportException: java.lang.NullPointerException
	at com.t_oster.visicut.model.graphicelements.svgsupport.SVGImporter.importSetFromFile(SVGImporter.java:375)
	at com.t_oster.visicut.model.graphicelements.AbstractImporter.importFile(AbstractImporter.java:37)
	at com.t_oster.visicut.model.graphicelements.GraphicFileImporter.importFile(GraphicFileImporter.java:88)
	at com.t_oster.visicut.VisicutModel.loadGraphicFile(VisicutModel.java:711)
	at com.t_oster.visicut.VisicutModel.loadFile(VisicutModel.java:490)
	at com.t_oster.visicut.gui.MainView.loadFileReal(MainView.java:1871)
	at com.t_oster.visicut.gui.MainView.access$5400(MainView.java:136)
	at com.t_oster.visicut.gui.MainView$61.run(MainView.java:1788)
Caused by: com.t_oster.visicut.model.graphicelements.ImportException: java.lang.NullPointerException
	at com.t_oster.visicut.model.graphicelements.svgsupport.SVGImporter.importSetFromFile(SVGImporter.java:152)
	at com.t_oster.visicut.model.graphicelements.svgsupport.SVGImporter.importSetFromFile(SVGImporter.java:370)
	... 7 more
Caused by: java.lang.NullPointerException
	at com.t_oster.visicut.model.graphicelements.svgsupport.SVGImporter.importSetFromFile(SVGImporter.java:133)
	... 8 more
```

## System information
* OS: `elementary OS 5.0 Juno 64 bit, based on Ubuntu 18.04.2 LTS`
* Java
```
openjdk version "11.0.4" 2019-07-16
OpenJDK Runtime Environment (build 11.0.4+11-post-Ubuntu-1ubuntu218.04.3)
OpenJDK 64-Bit Server VM (build 11.0.4+11-post-Ubuntu-1ubuntu218.04.3, mixed mode, sharing)
```
* Inkscape: `0.92.4`
* visicut: `1.8-92-gc695e590`

mgmax, 2019-09-15:
Could you please upload the example files somewhere?

Kr4cker, 2019-09-22:
Here is the example file: [https://gofile.io/?c=zFYYS6](https://gofile.io/?c=zFYYS6)
I saved it as an inkscape svg.

t-oster, 2019-09-22:
Please try with Java 8, visicut is not yet Java 11 capable

mgmax, 2019-09-22:
I think this bug is actually two problems:
- embedded pixel graphics cause Null Pointer error with Java11 - will be fixed by new svgsalamander library ( #525 )
- linked pixel graphics are not supported (and the error message is not very helpful) - I will create a followup bugreport

closed 2019-09-22 by t-oster

Kr4cker, 2019-09-22:
I uninstalled openjdk-11-jre with 
```shell=
sudo apt-get autoremove openjdk-11-jre:amd64 openjdk-11-jre-headless:amd64
```
openjdk-8-jre was already installed. So I tried to install VisiCut again with:
```shell=
sudo dpkg -i /path/to/visicut.deb
```
Then I got the info, that some depencies are missing, so I wrote the command
```shell=
sudo apt-get install -f
```
But then it installed again openjdk-11-jre.

Thanks @mgmax for fixing the bug =)

PR #525 (2019-09-22): Update svgsalamander library
Fixes #524

Update using /usr/share/java/svgSalamander-1.1.1.jar from Debian's libsvgsalamander-java 1.1.1+dfsg-3
commit c6359ea1 (2019-09-22): Update svgsalamander library

Fixes #524

Update using /usr/share/java/svgSalamander-1.1.1.jar from Debian's libsvgsalamander-java 1.1.1+dfsg-3