t-oster/VisiCut #527: Shifts due to slow speed

rsramek, 2019-09-26:
I'm trying to cut an about 45 centimeter slightly trapezoidal shape on an Epilog Helix Mini 24. Cutting the long diagonal line creates an offset that depends on the speed of cutting. I'm suspecting an arithmetic error is happening somewhere.

On the image are three versions of the same path (which should be closed). The top one is the correct cut at 100 speed, 500 frequency, the one below is slightly offset (~0.3mm for the end of a 45 degree line) and cut at 10 speed, 500 frequency, and the bottom one is offset at ~1.3mm, cut at 5 speed, 500 frequency. If I cut rectangles, I do not observe an offset.

[image]

mgmax, 2019-09-26:
To find out whether it is a firmware bug or a problem in VisiCut, can you please test if the Epilog Windows driver has the same problem?

rsramek, 2019-09-30:
It does not. It cuts the same piece at 5 speed without errors.

mgmax, 2019-09-30:
Huh. Then the following would be helpful:
- Find the most simple file for which the problem occurs (if possible, no bent curves)
- Attach the following files: (just make a ZIP file, you can then drag-and-drop it here in the comment field)
  1. "Print to File" with Epilog's printer driver
  2. "File - Export Laser Code" in VisiCut
  3. Original file (SVG or whatever)

rsramek, 2019-10-01:
Here is a zip with the example (simple triangle) and both exports. At speed 3 there are gaps between what should be connected vertices. Depending on the order of cutting the gaps can appear at different or multiple places.

[visicut.zip](https://github.com/t-oster/VisiCut/files/3675452/visicut.zip)

mgmax, 2019-10-06:
Your VisiCut example looks okay: `PD0,0,   2331,193,   1098,299,   0,0` (spaces added by me for readability). The coordinates are absolute coordinates so that must be a closed shape starting in (0,0) and ending in (0,0). The scale is also correct: 2331 = 400 dpi * 4.83 inch shape width. 

The result from the Windows driver only contains engraving, not cutting (no "PD" command). Please make sure to use a small enough line width and exactly the settings which you used for testing it on the real lasercutter.

If you are curious you can also look at the data dump yourself: `lib/LibLaserCut/util/dump-epilog-raw/dump-epilog-raw-print-file.py <filename>`.  `PD` is cutting, long unreadable lines with `*b` etc. are engraving.

mgmax, 2019-11-01:
@rsramek can you please send the file from the Windows driver that cuts correctly? Without, I cannot help.

rsramek, 2020-01-09:
Sorry for the long delay. As you suspected, this is indeed an epilog driver issue and not a visicut issue.
The epilog driver defaulted to engraving and ignored changes entered through Inkscape print menu, confusing me as to what it is doing. After changing the settings through windows settings and printing by saving and opening a pdf rather than directly from Inkscape I finally got it to vector cut, manifesting the same issue.
Thanks for the help!

closed 2020-01-09 by rsramek