t-oster/VisiCut #532: VisiCut crashes on importing settings with Java version 8

pallavagarwal07, 2019-10-15:
**Note: The problem was fixed using Java 11 instead of Java 8.**

Original Bug report:
Trying to download recommended settings (for any lab) on the latest build crashes on my PC.
I ran into this while trying to download settings that were added by me in a recent pull request (by the name of Makerspace BLR).

The error dump on terminal is not much helpful:
```
Oct 14, 2019 3:06:01 PM com.t_oster.visicut.managers.FilebasedManager readObjectFromXmlFile
WARNING: Could not load object in current XML format, retrying with old format from 2012:
Can't load settings.
'/usr/local/home/pallavag/.visicut' doesn't exist. We create it.
Copying default settings...
done.
Copying default settings...
done.
cannot find default settings folder for mappings
Copying default settings...
done.
Copying default settings...
done.
Generating some defaults if no examples exist...
Security framework of XStream not initialized, XStream is probably vulnerable.
Security framework of XStream not initialized, XStream is probably vulnerable.
Security framework of XStream not initialized, XStream is probably vulnerable.
Saving generated settings...
/usr/bin/visicut: line 10: 46852 Segmentation fault      java -Xms256m -Xmx2048m -jar "$DIR/Visicut.jar" --singleinstanceport $port "$@"
```

$ java -version
```
openjdk version "1.8.0_181-v7"
OpenJDK Runtime Environment (build 1.8.0_181-v7-270956655-270956655)
OpenJDK 64-Bit Server VM (build 25.181-b01, mixed mode)
```

OS: Debian Rolling Release (not off the shelf, so doesn't exactly have a name)

mgmax, 2019-10-20:
The Java runtime crashes ("Segmentation fault"). This is only possible if there is a bug in Java itself, so the problem is not with VisiCut. Debian oldstable has a newer Java8 version (8u222-b10-1~deb9u1), try updating to that.

I think this issue can be closed because it's not a bug in VisiCut.

pallavagarwal07, 2019-10-26:
You might want to put a small note in the README to help someone in the future. Closing issue since Java 11 works.

closed 2019-10-26 by pallavagarwal07