t-oster/VisiCut #540: Provide clear info about supported lasercutters

matkoniecz, 2019-11-07:
In my case I want to figure out whatever is supposed to be able to connect to ATMSolutions C02 PRO35 MINI, using Linux 16.04.

I started from https://github.com/t-oster/VisiCut README

Found http://visicut.org/ with 

> currently supporting the Epilog ZING, MINI and HELIX and Lasercutters with the LAOS board and some more

and I, laser newbie, have no idea whatever MINI in my laser is about the same as MINI on list of supported ones (second laser in the makerspace is larger ATMSolutions CO2 ATMS PRO1016 80 W - without MINI, what suggest to me that MINI is accidental, but I may be wrong - again, newbie to lasercutters).

Seems that https://github.com/t-oster/VisiCut/wiki/FAQ https://github.com/t-oster/VisiCut/wiki/Visicut-getting-started https://github.com/t-oster/VisiCut/wiki/VisiCut-manual https://github.com/t-oster/VisiCut/wiki/Welcome-to-VisiCut's-Documentation have no relevant help

"platform-independent tool" suggests that it supports Linux (but is not making it clear, some consider "supports latest Mac and Windows 10 and nothing else" or "works on latest Android and latest iPhone" as "platform-independent tool".

t-oster, 2019-11-07:
I agree. Feel free to improve the README and WIKI.
Plaform indepedned means it's based on Java (11) which runs at least on Windows, Linux, OS X (probably BSD?). Smartphones are not targeted right now, because the UI stuff is very different on Android (but much of the java stuff could work).

Some of the drivers are generic. I think we should add a list for each driver which devices are tested successfully with this driver and maybe a tutorial on how to figure out which driver may be working.

mgmax, 2019-11-07:
As the drivers come from LibLaserCut, we should avoid excessive redundancy and refer to https://github.com/t-oster/LibLaserCut/blob/master/README (which is, unfortunately, incomplete) for the full list.

t-oster, 2019-11-14:
No matter where we store and link this information, we should definitely have a list with manufacturer, model, driver and example settings for every device someone successfully hooked up with VisiCut. I get those questions a lot like "does the software support device xy?"

mgmax, 2019-11-17:
To have it editable easily, I created a wiki page: https://github.com/t-oster/VisiCut/wiki/Supported-Hardware. It is now linked on the Visicut website and wiki. I will soon update the LibLaserCut Readme, too. 
I chose to put it in the VisiCut Wiki because the example settings refer to VisiCut and not so much to LibLaserCut.

mgmax, 2019-11-17:
@matkoniecz I think you misunderstood the README. We support the Zing, Mini and Helix Models by the manufacturer Epilog. Most certainly your device is something different. From some googling it seems it's a RDWorks device. See https://github.com/t-oster/VisiCut/issues/404 for ongoing development by @fablabnbg and https://github.com/fablabnbg/visicut for beta releases for experimenting.

closed 2019-11-17 by mgmax