t-oster/VisiCut #543: Compiling in Netbeans

mrlinux1, 2019-11-20:
I set up Netbeans 11 on Linux Mint 19.2 and did all steps from the wiki to 
setup Visicut in Netbeans.

When I try to run the project I get the following error:

```
cd /home/tsgebert/NetBeansProjects/VisiCut; JAVA_HOME=/usr/lib/jvm/java-11-openjdk-amd64 /opt/netbeans/java/maven/bin/mvn "-Dexec.args=-classpath %classpath de.thomas_oster.visicut.gui.VisicutApp" -Dexec.executable=/usr/lib/jvm/java-11-openjdk-amd64/bin/java process-classes org.codehaus.mojo:exec-maven-plugin:1.5.0:exec
Scanning for projects...
                                                                        
------------------------------------------------------------------------
Building VisiCut 1.9-SNAPSHOT
------------------------------------------------------------------------
The POM for org.kabeja:kabeja:jar:0.4 is missing, no dependency information available
The POM for org.kabeja:kabeja-svg:jar:0.4 is missing, no dependency information available
The POM for org.kabeja:kabeja-xslt:jar:0.4 is missing, no dependency information available
The POM for de.thomas-oster:liblasercut:jar:1.9-SNAPSHOT is missing, no dependency information available
------------------------------------------------------------------------
BUILD FAILURE
------------------------------------------------------------------------
Total time: 0.323 s
Finished at: 2019-11-20T13:21:18+01:00
Final Memory: 8M/37M
------------------------------------------------------------------------
Failed to execute goal on project visicut: Could not resolve dependencies for project de.thomas-oster:visicut:jar:1.9-SNAPSHOT: The following artifacts could not be resolved: org.kabeja:kabeja:jar:0.4, org.kabeja:kabeja-svg:jar:0.4, org.kabeja:kabeja-xslt:jar:0.4, de.thomas-oster:liblasercut:jar:1.9-SNAPSHOT: Failure to find org.kabeja:kabeja:jar:0.4 in https://repo.maven.apache.org/maven2 was cached in the local repository, resolution will not be reattempted until the update interval of central has elapsed or updates are forced -> [Help 1]

To see the full stack trace of the errors, re-run Maven with the -e switch.
Re-run Maven using the -X switch to enable full debug logging.

For more information about the errors and possible solutions, please read the following articles:
[Help 1] http://cwiki.apache.org/confluence/display/MAVEN/DependencyResolutionException

```
I tried to find a solution but couldn't find the right way.

How can I fix this?

Kind regards,
Thomas

t-oster, 2019-11-20:
Please try to run `mvn initialize` from the command line first. We may have to change this.

t-oster, 2019-11-20:
or from netbeans right-click -> run maven -> goals... and enter `initialize`

mrlinux1, 2019-11-20:
Thanks for the fast answer! 
Running the `mvn initialize` in the projects folder solved the errors regarding kabeja libraries but not yet for de.thomas-oster:liblasercut:jar:1.9-SNAPSHOT:

`The POM for de.thomas-oster:liblasercut:jar:1.9-SNAPSHOT is missing, no dependency information available`

Do you have alos a fix for this? ;-)

Many thanks in advance!

t-oster, 2019-11-20:
Open LibLaserCut (should be a subfolder if you used git clone --recursive, otherwise just clone it from https://github.com/t-oster/LibLaserCut) and compile this first

t-oster, 2019-11-20:
(right-click clean and build)

t-oster, 2019-11-20:
feel free to update the wiki accordingly

mrlinux1, 2019-11-20:
Many thanks! 
And I already updated the wiki ;-)

closed 2019-11-20 by mrlinux1