t-oster/VisiCut #544: VisiCut for Mac OSX 1.9-2-gc3fd722f won't open.

npoles, 2019-11-20:
After extracting the .zip for Mac OSX (version 1.9-2-gc3fd722f), the application will not open. I have right-clicked to bypass the prompt re: the app being from an unidentified developer. On first attempt, the application icon bounces in the dock for a second or so and then disappears. On subsequent launches, the icon flashes very quickly and the disappears. I am running the  version of Java shown below (from `java --version`). I have also tried with the latest version of the JDK from Oracle and the results are the same. 
```
openjdk version "11.0.5" 2019-10-15 LTS
OpenJDK Runtime Environment Corretto-11.0.5.10.1 (build 11.0.5+10-LTS)
OpenJDK 64-Bit Server VM Corretto-11.0.5.10.1 (build 11.0.5+10-LTS, mixed mode)
```

phdoerfler, 2019-11-21:
a) Does Visicut support Java 11?
b) what happens if you launch it manually with `java -jar visicut.jar`?

npoles, 2019-11-21:
a) System Requirements on [download.visicut.org](https://download.visicut.org/) list "Java >= 11" as suitable Java versions.
b) VisiCut opens without a problem when launched manually with `java -jar Visicut.jar`

mgmax, 2019-11-21:
What is the output when launching from the commandline? See https://github.com/t-oster/VisiCut/wiki/Development:-Running-from-commandline

t-oster, 2019-11-21:
I guess I did not change the package of the main class in the osx info.plist . I will fix that tomorrow

t-oster, 2019-11-21:
Can you try 1.9-8?

npoles, 2019-11-22:
Same result unfortunately. 
This is probably irrelevant, but I've noticed that macOS labels the app version as $VERSION.

t-oster, 2019-11-22:
Now since we use Maven, we should migrate to https://github.com/federkasten/appbundle-maven-plugin for the osx build anyway.... I will report when I fixed it

t-oster, 2019-12-05:
@npoles I am trying to fix that and even bundle java with VisiCut. The latest version uses the universalJavaApplicationStub from https://github.com/tofi86/universalJavaApplicationStub/issues and includes a JRE, but it does not work on my virtual mac.
Can you test if this works A) without java installed and B) with java installed?

npoles, 2019-12-06:
Happy to give it a shot. Just to confirm, you want me to try 1.9-13, correct?

t-oster, 2019-12-06:
Yes. I tried it myself and it works after I installed java 11 from adoptopenjdk. However there is a JRE embedded, but it is not found. I am doing something wrong in the info.plist, but I could not yet figure out what.

npoles, 2019-12-06:
I just uninstalled Java and v1.9-13 seems to be working fine on my end.

t-oster, 2019-12-07:
Interesting.... Could you double check which Java process is running when running visicut? Should be doable eg with 'ps aux|grep java' in the terminal while visicut is running

npoles, 2019-12-09:
Looks to be the one in the application package. 

`/private/var/folders/sm/qwkh_21s4l90h_67cz_83d9c0000gn/T/AppTranslocation/D6C1D9B4-4DF1-4992-AB19-6E7B567614B4/d/VisiCut.app/Contents/Plugins/JRE/Contents/Home//bin/java -cp /private/var/folders/sm/qwkh_21s4l90h_67cz_83d9c0000gn/T/AppTranslocation/D6C1D9B4-4DF1-4992-AB19-6E7B567614B4/d/VisiCut.app/Contents/Java/* -splash:/private/var/folders/sm/qwkh_21s4l90h_67cz_83d9c0000gn/T/AppTranslocation/D6C1D9B4-4DF1-4992-AB19-6E7B567614B4/d/VisiCut.app/Contents/Resources/ -Xdock:icon=/private/var/folders/sm/qwkh_21s4l90h_67cz_83d9c0000gn/T/AppTranslocation/D6C1D9B4-4DF1-4992-AB19-6E7B567614B4/d/VisiCut.app/Contents/Resources/MacIcon.icns -Xdock:name=VisiCut -Dapple.laf.useScreenMenuBar=true -Xdock:name=VisiCut -Xms128m -Xmx1048m -splash:splash.png de.thomas_oster.visicut.gui.VisicutApp`

t-oster, 2019-12-09:
Yes it worked for me also in a fresh Mojave vm. So I assume this can be closed

closed 2019-12-09 by t-oster