t-oster/VisiCut #546: Required files to import settings should not contain personnal data or volatile settings

neomilium, 2019-11-30:
When you want to share your devices, materials and lasercut settings, the file `setttings/settings.xml` is currently required, which was not the case previously.

Unfortunately, this file contains personal data and volatile settings.
This have two side effects:
 * `recentFiles` entry could be a personal data disclosure
 * `windowBounds`, `lastMaterial`, and maybe more, do appears in a `git diff` when you have your shared `visicut-settings` managed through VCS.

IMHO, all entries which are personal and/or volatile are user-defined and should be moved in a dedicated file. This way the new file should be explicitly excluded from import and a recommendation to ignore it in Git could be made in Wiki.

Note: This discussion follow the one started here:
https://github.com/haum/visicut-settings/issues/1#issuecomment-559681125

closed 2019-12-04 by mgmax

mgmax, 2019-12-04:
@t-oster can you please check what's wrong with the download page? The new version isn't showing up, neither is the previous one.

@neomilium 
The issue should be fixed in VisiCut 1.9-12 or newer which should be available for download in the next days.
I updated the Wiki at https://github.com/t-oster/VisiCut/wiki/How-to-add-default-settings-for-your-lab , please test if that works.

t-oster, 2019-12-04:
@mgmax right now the build and publish process is not triggered by commits. I have to start it manually. Should be uploaded in a few minutes.

neomilium, 2019-12-05:
Awesome! I will try ASAP. Thanks

commit 0062f436 (2019-12-04): Split settings into public and private part

The settings/settings.private.xml file can be deleted to clear user-specific transient settings such as recent files or last material.
Then, VisiCut falls back to settings/settings.xml which contains all information except for these transient settings.
Note that to simplify the code, the non-private settings are saved to both files; settings.private.xml takes preference except if it its modification date is older than settings.xml.

If the settings repository is managed by version control (e.g., git), settings/settings.private.xml should be added to [... cut]
PR #547 (2019-12-04): Split settings into public and private part
This more or less fixes #546.

It's not the ideal solution which would be splitting the fields into two disjunct files, <del>because that is currently impossible due to a dependency deadlock (new NetBeans cannot edit old GUI, old NetBeans cannot edit Java11 code). I hope @t-oster will manage to sort that out some time, </del>but for now I propose the following workaround:

There are two files:
- settings.private.xml: all settings including local transient settings.
- settings.xml: all settings, but with local transient settings nulled out

VisiCut tr [... cut]