t-oster/VisiCut #549: Configurable position default

arthurwolf, 2019-12-08:
Currently, in the "Position" tab, the default is top left, for the reference point.
On our machines, due to how they are, users always have to manually switch this to lower left instead.
It would be great if there was a configuration option in the machine configuration that would allow setting a value for the default of the reference point.
Thanks !

mgmax, 2019-12-08:
Do your machines also have "Origin lower left" enabled? I think it would make sense to measure from bottom if the zero coordinate shown in VisiCut is at the bottom anyway.

arthurwolf, 2019-12-08:
Yes they are configured to have the origin in the lower left. I don't know
if it's better to have a separate configuration option for this, or to have
this default follow the origin configuration.

t-oster, 2019-12-09:
I think we should not have too much options and since most users have to move the stuff initially anyway, I think we should just link this behavior with the `origin lower left` option.

arthurwolf, 2019-12-10:
Yes, linking this to the origin parameter would be perfectly fine, probably
even easier for most users. I really doubt it will be common to need a
different value for this and the origin.

commit e4697c29 (2019-12-31): Position panel: Adapt default anchor to machine origin

Fixes #549
closed 2019-12-31 by mgmax