t-oster/VisiCut #550: Configurable focus multiplier

arthurwolf, 2019-12-08:
On more recent versions of Visicut, if a user selects material and thickness from the "Material" tool, then at the beginning of the Gcode file a Gcode is added that looks something like : 

G0 Z[thickness selected]

However, on our machines, this is in fact not the correct thing to do. We would like either to not have the G-code at all, or if we really need to have it, that it be configured so the focus point is put in the middle of the thickness of the material ( 4mm thickness, focus point 2mm bellow surface, so 2mm Z movement ).

I think an easy way to address this, would be to add a configurable "focus multiplier" in the machine configuration, where the Gcode would be : 

G0 Z[thickness selected * focus multiplier]

This way we can set it to 0.5 to do the "middle of thickness" thing, or set it to 0 to completely disable it ( possibly remove the Z movement Gcode entirely from the Gcode file if the focus multiplier value is 0 ? )

Thanks!

mgmax, 2019-12-08:
This currently works as intended. You can set the focus (Z distance from top of material) for each material. For engrave this is typically different than for cut.

Some drivers allow (e.g. LAOS) disabling the focus setting.

I agree there should be some default focus distance, as it's "middle of the material for cut, top for engrave" in 99% of all cases.

arthurwolf, 2019-12-08:
I'm not saying it's not working as intended, it's just in our case a way to set it to 1/2 or 1/3 automatically ( without the user having to do the math and set it by hand every time ) would be nice. I think a multiplier setting would do the job fairly well. Hope that makes it clearer.

arthurwolf, 2019-12-08:
Are you saying in the material XML file with all the materials and thicknesses there's a way to set focus ?

mgmax, 2019-12-08:
No, in the tab where you also set power and speed.

arthurwolf, 2019-12-08:
Yes, that means users need to manually do the math and set the focus every
time ( which they currently do, and it gets tiring very fast, and they are
annoyed ).

mgmax, 2019-12-08:
I think a default focus depth (given as percent of material depth) for each profile (engrave,cut,mark,...) would be a good solution. At least we need to distinguish between "default to top" and "default to half material", because some cutting profiles are only used for decorative marking, so always using half depth would be wrong. Also, the maximum focus range of each lasercutter must be considered.

t-oster, 2019-12-09:
I think you are mixing two things. First there is the `autofocus` feature, which just moves down before the whole job and up afterwards if you check the `focus is initially on the laserbed` checkbox.  The other one is some drivers having the possibility to change the focus as parameter of the profile like speed and power. If you save a profile for every material anyway, you have to do the math once.

arthurwolf, 2019-12-10:
So you are saying I can modify the Z movement distance in the "materials"
profiles with different heights ? Do you remember the value name to modify
? ( if not I can probably find it in existing/example profiles )

mgmax, 2020-02-19:
you can set "focus" at the same place where you set "power" and "speed". If the driver supports it, the field is shown. For some drivers "supports focus" must be enabled in the machine settings. (And there may be the bug that after changing it, the field only shows up for materials that have no profile yet. If that occurs please report it.).
Example Settings: "Germany Erlangen FAU FabLab", choose "Epilog Zing" lasercutter, material "Acryl" 3mm, load any file with a red path (e.g. circle-20mm-red.svg from examples), go to the laser settings tab, there you have "focus = -1,5" (mm).

mgmax, 2020-08-16:
Closing as a duplicate of #42

closed 2020-08-16 by mgmax