t-oster/VisiCut #551: Configurable default new file position.

arthurwolf, 2019-12-08:
When importing a new file, it is always automatically positioned in the top left of the machine. However, on our machines, this means users *always* have to move the file/object to the lower left of the machine instead.
It would be great if there was a configuration option that would allow defining which corner is the default position for new files.
Thanks!

mgmax, 2019-12-31:
Currently, VisiCut uses the position in the SVG file as default. The top-left corner of the page is the top-left corner of the lasercutter. This is a useful feature if you want to cut something that doesn't start at the corner, and also quite intuitive as long as the SVG page size matches the lasercutter print bed size.

What's your suggestion regarding this use-case?

t-oster, 2020-01-01:
If this feature is really wanted, it should maybe be tied to the lasercutter origin setting like #549

mgmax, 2020-08-16:
Closing due to lack of response -- It's not clear what the intended solution should be. With the current suggestion, we would no longer have "position on SVG page = position on VisiCut page", which would break some use cases.

closed 2020-08-16 by mgmax