t-oster/VisiCut #552: Laser settings mapping fine tuning

arthurwolf, 2019-12-08:
Currently, if I have a part with say multiple layers or colors. When I am in the "Mapping" tab, I can set two operations to "Engrave". But then in my "Laser settings" tab, if I want one of the Engraving operations to be at 90% speed, and the other one to be at 70% speed, I have no way to do this.
I would need to be able to configure laser settings not just by operation "type", but also by actual individual operation. ( hope this makes sense )
A solution is to create new operation "types" so users have several different "Engrave", so they can select different ones and give them different values, but this is very hacky and does not look good in front of our users. We really hope a more elegant solution could be found.
Thanks a lot !

t-oster, 2019-12-09:
I am not sure if I understand you correctly, but you seem to want two different profiles. I would suggest two profiles like `engrave deep` and `engrave flat` which can have different speeds even for different materials. The whole point of the profiles whas the user to specify the intended result (e.g. engraved deep) instead of the parameters (like 90% speed), becuase the parameters change with the material, the intention not.

arthurwolf, 2019-12-10:
My users want to engrave at many different powers, two isn't enough. I had
to explain on the phone to one how to create 13 different
profiles/operations. It's a very nasty hack and he was quite annoyed at the
process. It's fine if there is a default, but the users need to be able to
modify it on a more granular level, ideally.

mgmax, 2019-12-11:
Maybe "engrave 3d" is what you are looking for? In this mode, you the grayscale colors directly map to power (e.g. 30% gray = 30% of configured power).

arthurwolf, 2019-12-11:
Engrave3D is bitmap, this problem is around vector paths.

t-oster, 2019-12-12:
So what you want is basically have the option to not use a profile, but specify the laser-settings per selection group?
[image]
[image]

Well that is not yet easily possible. The option to emulate this is to just add a number of profiles like 'cut 1, cut 2, cut 3...', but you would still have to switch between the tabs.

I gues you would like to have the laser settings directly in the mapping tab where the profile column is now.

Since this workflow is completely different from VisiCut's intended workflow I am not sure how to integrate this without making the ui too complicated.

If you have ideas and implement it in a way it does not break existing workflows it's fine with me. Maybe a pseudo-profile 'custom' which triggers new behavior.

However, isn't it enough for you if you just create 20 profiles (custom1, custom2...) and ship them with your fablab's lasercutter settings? then you users do not have to create those profiles. And if you then provide a predefined mapping which maps `layer n` to `custom n` you only need to be in the lasersettings tab and have most of the behavior you want.