t-oster/VisiCut #553: Size circuit

arthurwolf, 2019-12-08:
A common feature on laser cutters is when the machine is ready to cut, you can press a button and it will move the head in a rectangle around the exterior size of the job, so users can check if the job they set up fits within their workpiece.
This is very hard to implement in a host/controller software as it would need to read the whole Gcode file and try to determine the boundaries. 
However, in Visicut ( at least for Smoothie users ) this would be fairly easy to implement: have a "check area" button next to the "execute" button. When the button is pressed, Visicut uses ethernet ( which it already supports ) to talk to the Smoothieboard and give it a few G0s, to make it go around the work area. It's just one HTTP request. This would be very helpful to our users, this is very often requested.
Thanks a lot!

mgmax, 2019-12-08:
Sounds doable and helpful for many other machines. We need to be careful regarding usability:
- one more button
- it needs to be very clear that this button actually moves the machine -- for example, "check area" may be misunderstood as some software-only operation

By the way, if you use a camera (e.g. with VisiCam) you don't need these manual checks most of the time, and it is a lot more intuitive.

t-oster, 2019-12-08:
I remember that this is already possible. You have to define a profile with 0% power and use the 'cut outline' feature.

arthurwolf, 2019-12-10:
> I remember that this is already possible. You have to define a profile
> with 0% power and use the 'cut outline' feature.
Yes, but that's really not convenient enough to do for every job as some
users do, nearly as a reflex. Would really need a button. It can be a
disablable option ( or be disabled if some info like the IP for the control
board is not given ) so the button is hidden if this isn't enabled.

mgmax, 2020-01-01:
duplicate of #181

closed 2020-01-01 by mgmax