t-oster/VisiCut #560: wrong version number in VisiCut 1.9-35-g98e949b3

suschizu, 2020-01-10:
hello!
I've downloaded version 1.9-35-g98e949b3 but finder shows $VERSION instead
[image]

t-oster, 2020-01-11:
Does the splash screen show the correct version?

t-oster, 2020-01-11:
I guess I somewhere forgot to update the info.plist....

closed 2020-01-12 by t-oster

t-oster, 2020-01-12:
Does it work with the latest version?

suschizu, 2020-01-12:
hey thomas! sorry for the late reply - and thanks - everything looks good now - great :)

[image]

suschizu, 2020-01-12:
hey @t-oster - sorry for the late reply - and thanks! everything looks good - great - no more confusion :)
[image]

commit 6537f945 (2020-01-12): Version string in info.plist should be VISICUTVERSION. fixes #560