t-oster/VisiCut #561: build failure - java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty -> [Help 1]

HarterHorst, 2020-01-18:
Hi, 

When I build the liblasercut I get the following error on macOS Catalina.

#### Error Message ####
Plugin org.apache.maven.plugins:maven-clean-plugin:2.5 or one of its dependencies could not be resolved: Failed to read artifact descriptor for org.apache.maven.plugins:maven-clean-plugin:jar:2.5: Could not transfer artifact org.apache.maven.plugins:maven-clean-plugin:pom:2.5 from/to central (https://repo.maven.apache.org/maven2): Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty -> [Help 1]

#### Config ####

openjdk version "11.0.6" 2020-01-14 LTS
OpenJDK Runtime Environment Corretto-11.0.6.10.1 (build 11.0.6+10-LTS)
OpenJDK 64-Bit Server VM Corretto-11.0.6.10.1 (build 11.0.6+10-LTS, mixed mode)

Product Version: Apache NetBeans IDE 11.2
Updates: Updates available
Java: 11.0.6; OpenJDK 64-Bit Server VM 11.0.6+10-LTS
Runtime: OpenJDK Runtime Environment 11.0.6+10-LTS
System: Mac OS X version 10.15.2 running on x86_64; UTF-8; en_GB (nb)
User directory: /Users/andi/Library/Application Support/NetBeans/11.2
Cache directory: /Users/andi/Library/Caches/NetBeans/11.2

#### Skill level ####

I totally new to NetBeans and therefore I have to clue what's wrong here. So if you have any idea what's wrong here, pls. describe on click-level what to do.

Thanks

t-oster, 2020-01-19:
There seems to be something wrong with your Java Keystore config. https://stackoverflow.com/questions/6784463/error-trustanchors-parameter-must-be-non-empty. Please check if you are really running the right Java version and if it is installed properly.
You can try to compile via terminal with just the 'make' command.

mgmax, 2020-04-04:
Closing because there was no reply.

closed 2020-04-04 by mgmax