t-oster/VisiCut #562: No engraving levels on latest version with Smoothieboard controller

arthurwolf, 2020-02-05:
Hello.

On the latest version ( 1.9-36 ), a customer can not get engraving to work ( they are about to try 1.9-16 now ).

They tried several files with the same result: All lines are fully engraved, at the exact same value, no change at all from beginning to end of line. It's as if the whole picture was just black pixels ( except it isn't ).

I am attaching the visicut file they setup, the two files used ( a png with the actual engraving, and a dfx for the perimeter/contour that isn't very relevant here ), as well as the resulting Gcode file.

Edit: it won't let me attach the files as they are not supported extension, uploaded it to our server, you can download it at: https://cloud.robotseed.com/index.php/s/q3QzHu8p11LkGZk

If you look at the Gcode file, you will see that it sets the power ( S parameter ) to S0.3 ( the right value for the 30% that was configured ) at the beginning of the line, but then does not change it for any of the following pixels, despite the image having different greyscale value, and the pixel positions actually being specified. It sets it to 0 at the end of the line.

Any  idea what is going on here ? Is there something we could possibly be doing wrong ? We have many other customers using visicut without trouble, but this seems to occur on all files for this new customer, and the main difference is they are using the latest visicut version.

Any help would be extremely welcome.

Cheers.

arthurwolf, 2020-02-05:
For reference/easy access, here is a random sampling of two random engraving lines : https://pastebin.com/ENbckzUP
You can see it is moving pixel by pixel, but not changing laser power values despite the fact it should.

arthurwolf, 2020-02-05:
A picture of the results. This is supposed to be a greyscale pixelated Mario.

[image]

arthurwolf, 2020-02-05:
I was not about to witness it personally, but the customer reported that the test with version 1.9-16 resulted in no engraving ( ie the laser not turning on when it was supposed to engrave ). I do not have access to the Gcode for that test.

mgmax, 2020-02-06:
until the bug is fixed, you can try the old 1.8 version from https://download.visicut.org/archive . It should be from just before the bug was introduced.

mgmax, 2020-02-14:
Will be fixed in 1.9-41. -- @t-oster can you please upload a new build?

closed 2020-02-14 by mgmax