t-oster/VisiCut #571: File export completed indicator

arthurwolf, 2020-03-14:
This is an issue our laser cutter users have reported when using Visicut.

Essentially ( on Linux, all versions I have tried, including most recent ), there is no indicator telling the user when the file has been succesfully exported ( File -> Export laser code ).

The user goes to export the laser code, if it's a pretty simple file, that happens instantly, and if the file is very complex ( say it's a bitmap engraving ) it takes some time. But the user can't know if the file has been exported and they can move on to use it, or if the file is still processing, and they need to wait.

I'm not sure what the correct way of handling this would be. Maybe have an information message display in the same area warnings and errors get displayed, that tells user the file was succesfully exported?

Any improvement of this would be very welcome by our users. I'm available to test anything that gets changed regarding this. Thanks !

t-oster, 2020-03-14:
Well during normal job execution there is the progress bar through the progress listener interface. Since exporting is basically the same, we should be able to use the same procedure

mgmax, 2020-03-14:
I'm working on it - should be rather easy.

arthurwolf, 2020-03-14:
> I'm working on it - should be rather easy.

Cool, thank you !

Yes, when exporting the progress bar isn't really being helpful, fixing
that would work as a solution to this issue I think.

mgmax, 2020-03-14:
I'm curious on your workflow - I thought "export laser code" is more a debugging tool than something for end users. But, as usual, anything will be used in a workflow somewhere - https://xkcd.com/1172/ :-)
Also, is engraving actually usable on boards without native raster commands, or just a toy?

arthurwolf, 2020-03-14:
Users export gcode files, and those are then fed to Octoprint for actual
cutting.

Opening an issue for "visicut direct upload to octoprint" is on my todo
list.

arthurwolf, 2020-03-14:
> I'm curious on your workflow - I thought "export laser code" is more a
> debugging tool than something for end users. But, as usual, anything will
> be used in a workflow somewhere - https://xkcd.com/1172/ :-)
> Also, is engraving actually usable on boards without native raster
> commands, or just a toy?
Sure, it's totally usable.
Visicut generates too verbose engraving Gcode to get to the max speeds,
lasergrbl for example generates much more compact gcode and allows for
higher speeds, but even with the verbose gcode, you can do nice engraving
work with visicut. This is another issue I'm supposed to open, the verbose
engraving gcode thing.

closed 2020-03-14 by mgmax

arthurwolf, 2020-03-17:
> Closed #571 <https://github.com/t-oster/VisiCut/issues/571> via 3c825c4

Awesome, thank you so much.

commit 3c825c4c (2020-03-14): Fix #571: Show progress while saving file

Now Visicut will show exactly the same progress bar and message for "Export Laser Code" as it does for "Execute Job".