t-oster/VisiCut #574: Mailing Lists?

t-oster, 2020-03-30:
Hi,
I get a lot of emails from users asking how to setup some specific lasercutter with VisiCut. So I thought about a way to create a community of users for sharing configurations and helping each other.
Since GitHub issues should be for developers, what about a VisiCut users mailinglist?
I know, mailinglists are somewhat old-school and I also tried something stack-overflow-like (qa.visicut.org), but that only got spammed.
What do you guys think and how should we make it public if we decide to start a mailinglist?
(I created visicut-users@freelists.org)

mgmax, 2020-04-04:
Do I understand correctly that this mailing list should be completely public?
Then we could also use GitHub. I see two possibilities:
- Use issues here with some label like "question".
   We can create appropriate templates so that the label is added automatically, for example: https://github.com/grote/Transportr/issues/new/
  Pro: The search function also searches for bug reports (however not in LibLaserCut)
  Contra: You can't choose to only be notified for issues but not for questions.
- Create a second repository "visicut-questions" and use that for discussion.
  Pro/Contra: more or less like a mailing list with public archive, but on the same platform as the rest of development.
  Pro: Real issues can still be moved over to VisiCut with one click.

mgmax, 2020-04-04:
If you want to try out issue templates, here's a test repository with issue templates that automatically add labels (even if you don't have permissions to set labels, the template labels are still set automatically): https://github.com/mgmax/test/issues/new/choose

t-oster, 2020-04-04:
The only downside is that users have to create a GitHub account.

t-oster, 2020-04-04:
And they would have to subscribe to the issues in order for them to get notified about new questions.... I thought mainly about a users help other users scenario

mgmax, 2020-05-13:
GitHub discussions (public beta soon :tm:) could be a solution: https://github.blog/2020-05-06-new-from-satellite-2020-github-codespaces-github-discussions-securing-code-in-private-repositories-and-more/#discussions

mgmax, 2020-11-21:
It seems that one can request beta access to GitHub discussions in this forum post: https://github.community/t/can-one-apply-for-github-discussions-beta/121120

For a real-life example of GitHub Discussions, see https://github.com/nodejs/node/discussions

t-oster, 2020-12-15:
here it is https://github.com/t-oster/VisiCut/discussions/595, no idea if this is usefult.

closed 2020-12-15 by t-oster