t-oster/VisiCut #576: Win7 Visicut not launching

arthurwolf, 2020-04-29:
A Fablab installed 1.9-41 on all of their computers.
On Win10Pro computers, everything works fine.
On Win7Pro, Visicut now won't launch at all, it stays blocked on the loading screen ( which shows the version, etc ).
They then tried uninstalling and re-installing an older version (-35 and -36 were tried ), and same issue ( not going beyond loading screen ).
They are able to get it to launch by selecting right click / solve compatibility issues / win X3 service pack 3 compatibility. Then Visicut launches. But it doesn't launch when normally launching it. And they don't know how to make the compatibility thing permament.

Any idea what could be going on?

Thanks!

t-oster, 2020-04-29:
Can you try to launch via CMD.exe, maybe it shows what's happening. It should use the bundled java version, so I guess you have no java installed on the machine?
Maybe the singleinstanceport thing is making problems....

mgmax, 2020-05-02:
Win7 is "end of life", except if you pay 50€ or so per year for extended support. Either you should upgrade ASAP for security, or if you have the money for extended support then you should consider donating some to Thomas so that he fixes the problem :-D

arthurwolf, 2020-05-02:
I don't have this problem, a non-profit Fablab has this problem. I'm
helping them with the issue mostly because it'd be a hassle for them to try
to explain the issue in English.
I don't know if they pay for extended support or not and frankly I don't
think it's any of my business.
A lot of fablabs around here use Win7, often they use donated hardware on
which I'm not sure Win10 would run.

mgmax, 2020-05-02:
Personally, I don't want to invest time in supporting insecure end-of-life operating systems. Win10 and Ubuntu run quite well on surprisingly old machines.
Meanwhile, you can use the old versions from https://download.visicut.org/archive and https://github.com/t-oster/VisiCut/releases .

t-oster, 2020-05-02:
First I would try to install Java 11 and confirm it runs on win7. Then try to launch the jar from the platform independent version

arthurwolf, 2020-05-02:
I've asked them to do that, their schedule is disturbed by the current
events, but I'll tell you the results as soon as I have them, thank you
very much for the help.

mgmax, 2020-08-15:
Closing because Windows 7 is no longer supported by Microsoft, same as Windows XP / 98. You can try using the platform independent ZIP version with your own Java 11 installation.

closed 2020-08-15 by mgmax