t-oster/VisiCut #578: Download page offline - migrate to GitHub releases?

t-oster, 2020-05-13:
My server is down and the company is not answering.
Maybe this is n opportunity to migrate to CI automatically create a GitHub release for every tagged commit (https://docs.travis-ci.com/user/deployment/releases/).
I will try that tomorrow.
However I am not sure if there are any limitations on the number of releases or the size of the build artifacts....

mgmax, 2020-05-13:
If it's too hard with travis, GitHub actions may be a viable alternative.

t-oster, 2020-05-14:
@mgmax seems to work now. You just have to tag a commit and push it to master to trigger a release. I would recommend to use 1.9-xx as tag just counting upwards.

t-oster, 2020-05-14:
download links should be updated soon

closed 2020-05-14 by t-oster