t-oster/VisiCut #584: Failed to execute goal org.apache.maven.plugins:

gatonero, 2020-06-18:
While compiling VisiCut I get following error:
`[MVNVM] Using maven: 3.6.3
[INFO] Scanning for projects...
[INFO] 
[INFO] --------------------< de.thomas-oster:liblasercut >---------------------
[INFO] Building liblasercut 1.9-SNAPSHOT
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- jacoco-maven-plugin:0.8.5:prepare-agent (agent) @ liblasercut ---
[INFO] argLine set to -javaagent:/home/cs/.m2/repository/org/jacoco/org.jacoco.agent/0.8.5/org.jacoco.agent-0.8.5-runtime.jar=destfile=/var/tmp/pamac-build-cs/visicut-git/src/VisiCut.git/LibLaserCut/target/jacoco.exec
[INFO] 
[INFO] --- maven-resources-plugin:2.6:resources (default-resources) @ liblasercut ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] Copying 2 resources
[INFO] 
[INFO] --- maven-compiler-plugin:3.1:compile (default-compile) @ liblasercut ---
[INFO] Changes detected - recompiling the module!
[INFO] Compiling 79 source files to /var/tmp/pamac-build-cs/visicut-git/src/VisiCut.git/LibLaserCut/target/classes
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  1.527 s
[INFO] Finished at: 2020-06-18T22:59:00+02:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.1:compile (default-compile) on project liblasercut: Fatal error compiling: invalid target release: 11 -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
make: *** [Makefile:28: libLaserCut] Fehler 1
==> FEHLER: Ein Fehler geschah in build().
    Breche ab...`

My system is:
`Operating System: Manjaro Linux 
KDE Plasma Version: 5.18.5
KDE Frameworks Version: 5.70.0
Qt Version: 5.15.0
Kernel Version: 5.4.44-1-MANJARO
OS Type: 64-bit
Processors: 12 × AMD Ryzen 5 2600 Six-Core Processor
Memory: 15,7 GiB
`

t-oster, 2020-06-19:
looks as if you are compiling with java 8, we need at least java 11

gatonero, 2020-06-19:
Thanks a lot. Perfect Solution :-)

closed 2020-06-19 by t-oster