t-oster/VisiCut #585: Request: Support for Duet based CNCs (Workbee)

FelixHauser, 2020-06-27:
Hi there,

I have ordered a Laser module as an upgrade to my Duet based CNC, aka Workbee (Ooznest, Openbuilds, etc.). To my surprise finding a workflow to generate Gcodes to send to the Duet is not an easy task. That's why I'd like to request support for this environment. The Duet boards run on RepRap Firmware.

I tried to set up a generic Gcode machine, but I'm not sure it would work (laser module not yet there). I don't see any Laser on-Laser off code. If I check the checkbox "Force Laser Off During G0 Move"  the generated code looks like this:

```
G28
 G21
 G90
 M3
G0 X242.620800 Y158.597600 F1500 S0
G1 X247.599200 Y158.597600 F500
G1 X251.510800 Y158.597600
G1 X256.489200 Y158.597600
G0 X257.860800 Y158.648400 F1500 S0
G1 X252.882400 Y158.648400 F500
G1 X246.837200 Y158.648400
G1 X241.858800 Y158.648400
G0 X241.503200 Y158.699200 F1500 S0
G1 X246.481600 Y158.699200 F500
G1 X253.796800 Y158.699200
G1 X258.775200 Y158.699200
G0 X259.486400 Y158.750000 F1500 S0
G1 X254.508000 Y158.750000 F500
G1 X246.176800 Y158.750000
G1 X241.198400 Y158.750000
G0 X240.893600 Y158.800800 F1500 S0
G1 X245.872000 Y158.800800 F500
G1 X255.117600 Y158.800800
```

There is a S0, but I've never seen a S command by itself

Thanks in advance

t-oster, 2020-06-28:
Does your selected profile have a power setting > 0?

FelixHauser, 2020-06-28:
> Does your selected profile have a power setting > 0?

I think I was misunderstanding something Gcode related, since I never worked with lasers before. I will experiment a bit right now and report back. Thanks.

mgmax, 2020-08-15:
Closing as there was no reply. Feel free to answer if you have new input.

closed 2020-08-15 by mgmax