t-oster/VisiCut #586: Inkscape extensions

LenoreHorner, 2020-08-13:
The wiki says "In VisiCut, select Extras -> Install Inkscape extension... to install the extension to Inkscape. Then you can restart Inkscape, open a file, Extensions -> Lasercut -> Open with VisiCut." which I have done before but in the 1.9-70-g2c0db535 version (at least the Mac version), there isn't an Extras menu and the Actions menu is empty.  How do I install the extension?

LenoreHorner, 2020-08-13:
I downloaded the general version (not all just no label zip file) and the Extras menu is there.  Ran that.  Now it works.  So, still: don't use the Mac version just use the generic version.

t-oster, 2020-08-13:
It looks like the packaging for OSX needs to be improved. The extensions should be definitively installable via the OSX package, so this should be fixed.
My OSX VirtualBox is not working currently, so has any OSX developer time to look into this?

sebastian-schmid, 2021-09-18:
This Issue is still present. I tried it with the latest version of visicut. The "Action" Menu is just not shown, all other Menu Options are working. I have no idea so far why this happens.

[image]

[image]

mgmax, 2021-09-18:
Here's my guess from the source code:
GetVisicutFolder() returns the path of the JAR file. IsInkscapeExtensionInstallable() looks for ./inkscape_extension/ inside that path.

https://github.com/t-oster/VisiCut/blob/9e819d34b501d015cc9fe7bc214ca3516db410b7/src/main/java/de/thomas_oster/visicut/misc/Helper.java#L365

https://github.com/t-oster/VisiCut/blob/9e819d34b501d015cc9fe7bc214ca3516db410b7/src/main/java/de/thomas_oster/visicut/misc/Helper.java#L692

However, in the MacOS package, the file layout is different: the JAR file is in VisiCut.app/Contents/Java, while the inkscape extension is in VisiCut.app/Contents/Resources/inkscape_extension.

So GetVisicutFolder() should be modified: If GetvisicutFolder() /examples/ exists. If not, try again with ../Resources/Java/examples/ . Return the path that works.

This should also fix #502 .

Would be cool if you could help -- I can not because I don't have a Mac.

sebastian-schmid, 2021-09-18:
As far as I understand the code now I think this is not fixable within getVisicutFolder() function cause this function does not know about subpaths because no arguments are passed to that function it just returns the Visicut Path.

So I think this check has to be implemented in isInksacpeExtensionInstallable() and in installInksapeExtension(). There we could just use isMacOS() function to get the right path straight away?

Do I miss something or did you think something else I misunderstood?

mgmax, 2021-09-18:
The /examples/ folder is always there, so you can just check for that one in getVisiCutFolder():
```
folder = folder of .jar file (as before)
if exists(folder + "/example") {
   return folder;
} elseif exists(folder + "../Resources/Java/example)  {
   return folder + "../Resources/Java/example
} else {
   warning("Can not find visicut folder. Please report this bug.");
   return folder;
}
```
 As far as I understand the code, the purpose of getVisiCutFolder is to get the main folder that contains the subfolders inkscape_extension, illustrator_script, examples, et cetera.

The macOS app package is the only one with a special directory structure; everything else (ZIP, Windows EXE, Debian package, Arch package) has the normal structure. isMacOS() will also return true for the ZIP file version, so it is not the correct check here.

sebastian-schmid, 2021-09-19:
Thank you for your help to get me a step forward. I have at least a partly working dev build now which shows the menu on MacOS using the MacOS bundle.
It also says "Installed successfully" but Extension does not show up in inkscape afterwards and is not present in the user extension folder of inkscape on MacOS which is by default `/Users/USERNAME/Library/Application Support/org.inkscape.inkscape/config/inkscape/extensions`

Function installInkscapeExtension() does install it into `/Users/USERNAME/.config/inkscape/extensions` 

I think in installInkscapeExtension() we need to check if it's on Mac and handle it different like for windows already in that function or do you see another solution for this?

Tested on MacOS 10.15 with inkscape 1.1

sebastian-schmid, 2021-09-20:
Got the extension installation work by setting the MacOS path. But clicking "Open in VisiCut" in Inkscape Extensions does not open VisiCut. No Error shown. Not sure if some path in the extension scripts is wrong or something different. I will try to debug this somehow.

Any ideas appreciated how to debug this.

mgmax, 2021-09-20:
Maybe there is no error message. Maybe the message is not visible. Maybe the extension is not called at all -- do any other extensions work?

At first you could check if the extension can show error messages to the user.
Just add `print("hello", file=sys.stderr); sys.exit(1)` to visicut_export.py in your Inkscape extensions directory and see if you get an error message.

sebastian-schmid, 2021-09-21:
Thank you, that is working so extension is called by inkscape now. Will try to find whats going wrong within the extension.

Tried a few things and looks to me that it's just crashing on `daemonize.createDaemon()` no idea why at the moment. Maybe something MacOS / python related, any idea?

If I remove this line, looks like it should work then but blocks inkscape, not a solution but gets me to the cmd execution. This fails because the path to Visicut.jar ist wrong here `/Applications/VisiCut.app/Contents/Resources/Java/Visicut.jar` has to be `/Applications/VisiCut.app/Contents/Java/Visicut.jar`.

That seems to be wrong in `VisiCut.MacOS` file because there is `$DIR/Visicut.jar` but `$DIR` is the wrong path here cause the .jar file is not in the same directory. Fixed this.

Now I am ending up with a Java Rutime Error "class file version 55.0) this version of the Java Runtime only recognizes class file versions up to 52.0" not sure if this will be on all MacOS because I may not be on defaults here. Will see if I can test this on a default  MacOS Installation.

t-oster, 2021-09-21:
You need to call the java binariy from inside the visicut.app directory in order to make sure you have the correct version (and java available at all)

mgmax, 2021-09-21:
Some stupid questions from someone who does not use MacOS:

- How can it be that `VisiCut.MacOS` does not work but "starting VisiCut standalone" works?
- Is there a new launcher script that replaced the old `VisiCut.MacOS`?
- Do the instructions on https://github.com/t-oster/VisiCut/wiki/Development:-Running-from-commandline no longer work?

sebastian-schmid, 2021-09-21:
Wondered also about that. I think on a standalone start it uses `/Applications/VisiCut.app/Contents/MacOS/universalJavaApplicationsStub` is that correct? @t-oster 

This script does a lot more than the `VisiCut.MacOS` one.

On MacOS I am not able to start visicut by typing visicut to the Terminal. "Command not found."

t-oster, 2021-09-21:
Yes, see https://github.com/t-oster/VisiCut/blob/master/distribute/mac/VisiCut.app/Contents/Info.plist
basically it runs universalJavaApplicationsStub and sets JAVA_HOME to Contents/Plugins/JRE/Contents/Home/ and adds a bunch of parameters.
How about you try to run it with "open /path/to/VisiCut.app"? Maybe this runs it correctly.

t-oster, 2021-09-21:
@mgmax in OSX a folder with .app ending acts as a self-contained application which is shown as a single file and can be moved around. The info.plist file contains all necessary metadata. It is totally different from the zip file which just runs a bash script.

sebastian-schmid, 2021-09-21:
`open /Applications/VisiCut.app` works, but looks like it's not possible to pass the arguments this way to VisiCut.

`open /Applications/VisiCut.app "$@"` does not work or it works and there is another problem with the file, path or conversion.

sebastian-schmid, 2021-09-21:
I think we need a bash script like the Linux one, setting Java to be used etc. or do you see an easier solution?

t-oster, 2021-09-22:
https://superuser.com/a/801768 says open has an --args option. So it must be **open -a /path/to/visicut --args "$@"** ? Or even with the -b option?

t-oster, 2021-09-22:
If I understand correctly we should use the -b option so we don't need absolute paths

mgmax, 2021-09-22:
I would suggest that the Inkscape extension still calls `VisiCut.MacOS`, which then opens the app. Then, the Inkscape extension would remain as simple as possible. This will save effort for other extensions  (Illustrator and possibly more in the future).

Pseudocode for `VisiCut.MacOS`
```
if the file .../VisiCut.app exists then
    open .../VisiCut.app
else 
   do whatever the script did before (for the ZIP file version)
end if
```

sebastian-schmid, 2021-09-22:
Thanks for your feedback, just pushed this including your suggestion to the pull request.

PR #620 (2021-10-17): Macos inkscape extension install fix
This Pull Request will fix #586 and #502 

Would be also great if someone can test on windows to make sure this is not breaking something.
closed 2021-10-17 by t-oster