t-oster/VisiCut #589: NullPointerException when tryng to run 1.8_107 on Manjaro

JozefAB, 2020-09-15:
I am running Manjaro (5.4.64-1-MANJARO) and trying to use Visicut 1.8_107. I have openjdk 7,8,11 & 14 installed and have tried every one to get it to work. I need to use an older version as the latest Visicut version will not work with the laser at the hackspace I use. I have installed 1.8 without issue but when I try to run it for the first time it fails with the following error

Error: Application class com.t_oster.visicut.gui.VisicutApp failed to launch
org.jdesktop.application.Application$1.run(Application.java:177)
java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:316)

This is the error log from terminal

WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.thoughtworks.xstream.core.util.Fields (file:/usr/share/visicut/lib/xstream-1.4.11.1.jar) to field java.util.TreeMap.comparator
WARNING: Please consider reporting this to the maintainers of com.thoughtworks.xstream.core.util.Fields
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Security framework of XStream not initialized, XStream is probably vulnerable.
Security framework of XStream not initialized, XStream is probably vulnerable.
Security framework of XStream not initialized, XStream is probably vulnerable.
Security framework of XStream not initialized, XStream is probably vulnerable.
Security framework of XStream not initialized, XStream is probably vulnerable.
Sep 15, 2020 9:45:47 AM org.jdesktop.application.Application$1 run
SEVERE: Application class com.t_oster.visicut.gui.VisicutApp failed to launch
java.lang.NullPointerException
	at com.t_oster.visicut.gui.MainView.refreshRecentFilesMenu(MainView.java:518)
	at com.t_oster.visicut.gui.MainView.<init>(MainView.java:388)
	at com.t_oster.visicut.gui.VisicutApp.startup(VisicutApp.java:77)
	at org.jdesktop.application.Application$1.run(Application.java:171)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:316)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
	at com.t_oster.visicut.gui.VisicutApp$1EventQueueProxy.dispatchEvent(VisicutApp.java:131)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

java.lang.Error: Application class com.t_oster.visicut.gui.VisicutApp failed to launch
	at org.jdesktop.application.Application$1.run(Application.java:177)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:316)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
	at com.t_oster.visicut.gui.VisicutApp$1EventQueueProxy.dispatchEvent(VisicutApp.java:131)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: java.lang.NullPointerException
	at com.t_oster.visicut.gui.MainView.refreshRecentFilesMenu(MainView.java:518)
	at com.t_oster.visicut.gui.MainView.<init>(MainView.java:388)
	at com.t_oster.visicut.gui.VisicutApp.startup(VisicutApp.java:77)
	at org.jdesktop.application.Application$1.run(Application.java:171)
	... 14 more

Any advice on how I could get this running would be appreciated.

t-oster, 2020-09-15:
Can you verify that if you run java -version that 11 is active? And then try to run the jar with Java -jar. Maybe you need to move your .visicut folder to check if invalid settings are the problem

JozefAB, 2020-09-15:
Hi Thomas, thank you for getting back to me so quickly. It was the settings folder that was causing the problem. I moved it, opened the jar file again and it worked. Thank you for your help!

mgmax, 2020-09-15:
Could you explain why you can't use the newer version? I don't know of any driver that we removed. What machine are you using?

JozefAB, 2020-09-15:
Hi Max. I am not aware of the details I’m afraid. I have just been told that I need to use an older version of visicut because a different version of Laos needs to be compiled to get the newer version working with the cutter we have at the hackspace.

t-oster, 2020-09-16:
But as I understand correctly this is resolved?

mgmax, 2020-09-16:
Let's continue the discussion at https://github.com/t-oster/LibLaserCut/issues/131

closed 2020-09-16 by mgmax

JozefAB, 2020-09-16:
Thanks to both of you. I’ll keep an eye on the thread in the link. I think that is the problem we’re having with the new version