t-oster/VisiCut #594: Get rid of splash screen generation

phdoerfler, 2020-11-21:
Instead of invoking bash and a bunch of other command line programs to replace text in the splashsource svg and then convert ist to PNG, I propose that instead we:

- Generate the required PNGs once and commit them; or
- generate the PNGs during the build using JVM libraries; and
- do not include version information in the png file but instead use the `SplashScreen splash = SplashScreen.getSplashScreen();` functionality to draw the version number and any other changing information onto the splash screen image during startup. We might as well use this to indicate which part of Visicut is currently loading.

This will make it a lot easier to build and package visicut on different platforms, not requiring sed, rm, rsvg-convert, bash, echo, cat, and so on.
It will also improve the user experience by providing more detailed startup information.
It further improves the user experience by using the java font rendering instead of whatever rsvg-convert is using.

Sounds good? If there aren't any objections I ma start on this right away.

t-oster, 2020-11-21:
Hi @phdoerfler , yes the variant with drawing the version number onto the spash screen sounds good.
Committing a png is not so easy, because the commit itself would change the version number, because the commit-id is part of the version number.

mgmax, 2020-11-21:
I'm always in favor of removing unnecessary dependencies, but I am unsure how much this helps with regard to "building on different platforms":

There is a fallback mechanism, so it is already possible to build and even package without rsvg-convert. The only differences are:
- You see "Version: snapshot" (the value that is hardcoded in splash-fallback.png) instead of the correct version number.
- The high-DPI variants (splash-2x.png, -3x.png) of the splash image are missing, so the splash may be smaller or slightly blurred.

The other programs you mentioned (sed, rm, echo, ...) are required by almost every piece of the packaging process, see distribute.sh. I don't see a practical alternative to get rid of them. Building (without packaging) should already be possible without any of this.

If your goal is to get rid of rsvg-convert, then please go ahead with any reasonable solution you like. We certainly have SVG rendering libraries on board, maybe you can even call these directly from the command line to convert SVG to PNG. Less beautiful but still okay would be to include all splash PNGs in the repository. 

If your goal is to build and package only with Maven + JDK, then there are many big issues to be solved, while the splash screen is only a really minor one.

t-oster, 2020-11-21:
@MaxGaukler is right, but I still like the idea of programmatically show text and loading status on the splash screen

phdoerfler, 2020-12-01:
That's all valid points and yes, my goal is a big one :) I've done most of the work already, the last thing is the few shell scripts and I shall start with the splash screen.