t-oster/VisiCut #597: Visicut + ruida controller

ARKopp, 2021-01-07:
I am thinking to get a chinise laser with a ruida controller 6442S. Does Visicut work with this controler?

mgmax, 2021-01-07:
Currently not. There is some ongoing development for certain Ruida controllers, you could possibly build upon that: https://github.com/t-oster/VisiCut/issues/404 https://github.com/t-oster/LibLaserCut/issues/105 https://github.com/t-oster/LibLaserCut/pull/134

closed 2021-01-07 by mgmax

ARKopp, 2021-01-07:
but this would do the shop right ? https://github.com/jnweiger/inkscape-thunderlaser

mgmax, 2021-01-07:
I don't know. It is supposed to be for the same lasercutter as the VisiCut driver under development. Not all Ruida boards are the same, so it may or may not work with the board you asked about. I would guess that smaller modifications may be required.

ARKopp, 2021-01-07:
i will ask the developers if it works with the ruida board i want to get