t-oster/VisiCut #601: Non-free icons

mgmax, 2021-01-24:
Looking at the license file
https://github.com/t-oster/VisiCut/blob/b7a98fee966a6adf0436d9f6005a6bda83b7f867/LICENSE#L5-L8
we currently have a number of icons with non-free or unclear licenses, including our main logo that is also used for the splash screen. These should be replaced.

t-oster, 2021-01-25:
suggestions welcome

mgmax, 2021-01-28:
First design ideas for splash + logo:

Grey:

[image]

[image]

Blue:

[image]

[image]

I'm not super happy with it; the proportions and the laser effects can definitely be improved. Hope that someone can do it better :-D

t-oster, 2021-01-29:
I like the blue version

phdoerfler, 2021-02-16:
I like the grey version slightly better ^^'

sebastian-schmid, 2021-10-16:
I also like the splash screen in grey more and the blue logo in grey would be a good match.

mgmax, 2022-12-03:
Could someone with a Mac please update these files
distribute/mac/MacIcon.icns
distribute/mac/VisiCut.app/Contents/Resources/MacIcon.icns
using the data from
icon*.png
?
thanks.

(I don't know why we have the file twice; probably the first one can be deleted.)