t-oster/VisiCut #609: .gcode extension missing when exporting a file

danke-0x21, 2021-05-26:
Hi,

When using the "Export to G-CODE" menu (my UI is not in english, the menu name may be different), you have to manually add the .gcode extension to the file name.
For instance, saving a file as PLF automatically add the .plf extension to the file name.
Not a big deal, but you know... workflows are sensitive cases... :)

mgmax, 2021-05-27:
I understand what you want, but the problem is that there are many different file endings for G-Code (.gcode, .ngc, .nc, .txt), and this also depends on the machine format (GCode / HPGL / whatever). So there is no easy fix. This would need to be a property of the specific lasercutter class or even a configuration option.

t-oster, 2021-05-27:
I would not make it that complex. The file chooser should have a drop-down for the type where we specify .gcode (default), .txt and any. If the user enters the filename with suffix, we don't change it. If the suffix is missing we append what ever is selected in the drop-down.

danke-0x21, 2021-05-27:
I didn't know about all the suffixes used for the G-Code files.

Both solutions (option in lasercutter profile and drop down menu) have pro and cons.
On the first one can be set for each lasercutter and then you don't bother anymore but you can't change it on the fly if needed.
On the second one you don't bother if you use default .gcode, but people who needs another suffix won't see the benefits.

So both are good for me since I'm a .gcode suffix user. I can't tell for other people.
On the other hand I have no programming skill so I can't help on which is the easiest to add.