t-oster/VisiCut #613: Missing icons in Ubuntu 21.04

davedarko, 2021-08-28:
We recently moved from Windows to Ubuntu for our laser cutting computer at the xHain Hackspace. For some reason some icons, probably GTK related, are missing. Starting it from the terminal it was complaining about a missing module. Installing the "gtk canberra module" did not change the appearance. Changing icon themes with gnome-tweaks neither. Not sure what I should do next.

[image]
[image]

mgmax, 2021-08-30:
Does switching to Java 11 work for you? (See #614)

mgmax, 2021-09-01:
Should be fixed in the next version (1.9-108 or so).
@t-oster can you please upload the new version?

t-oster, 2021-09-01:
2min

closed 2021-09-03 by mgmax