t-oster/VisiCut #621: Inkscape extension doesn't connect with visicut

Jorgininho, 2021-09-21:
Hello,

I'm working on a fresh install of inkscape and visicut, but the export from inkscape to visicut via extension menu doesn't seems to be working.
Here is the configuration : 
OS : Ubuntu 20.04.3 LTS
Inkscape : 1.1
Visicut : 1.9-102

When I try to export a design from inkscape to visicut, i get this : 
```
Traceback (most recent call last):
  File "visicut_export.py", line 267, in <module>
    VISICUTBIN = which("VisiCut.Linux", [VISICUTDIR, "/usr/share/visicut"])
  File "visicut_export.py", line 118, in which
    raise Exception("Cannot find executable {0} in PATH={1}.\n\n"
Exception: Cannot find executable 'VisiCut.Linux' in PATH=['/usr/share/visicut', '/usr/share/visicut', '/snap/inkscape/9090/bin', '/snap/inkscape/9090/usr/sbin', '/snap/inkscape/9090/usr/bin', '/snap/inkscape/9090/sbin', '/snap/inkscape/9090/bin', '/usr/local/sbin', '/usr/local/bin', '/usr/sbin', '/usr/bin', '/sbin', '/bin', '/usr/games', '/usr/local/games', ''].

Please report this bug on https://github.com/t-oster/VisiCut/issues

For a quick fix: Set VISICUTDIR and INKSCAPEDIR in /home/cervin/.config/inkscape/extensions/visicut_export.py

```

I don't see how to apply the quick fix solution, since the executable are indeed present in the mentioned folders

Thanks

mgmax, 2021-09-22:
My guess is that you have installed Inkscape as a "snap package". As far as I understand, snap packages are very locked down and can not call outside applications. As far as I understand, you even require special permission to call one snap application from another, and there is no way to launch non-snap applications -- not the best design...

The extension should at least be fixed to include a warning if "/snap/..." is on the PATH and Visicut can not be found.

You can try two workarounds:
- Start Visicut and then call the extension. If Visicut is already running, it will be contacted via network, maybe this is allowed in snap.
- Uninstall inkscape and re-install it from the normal "deb" packages (for example, using `sudo apt install inkscape` in the terminal window). However, you will then probably have an older Inkscape version.

mgmax, 2022-08-12:
A newer VisiCut version is available via https://download.visicut.org/ .

Please try the basic case of installing Inkscape via the standard way (technically as a DEB package via APT), not anything special like Appimage or snap.

I'm not sure why your Inkscape finds the extension menu entry (.inx file) but not the corresponding .py file. You could have a look at the user and system extension directory, if the files are there, and if the content is readable with some text editing program. These paths are shown in Inkscape: Edit - Preferences - System - (User) extensions.

Jorgininho, 2022-08-24:
So I've installed Inkscape vi command line
```
sudo add-apt-repository ppa:inkscape.dev/stable
sudo apt update
sudo apt install inkscape
```

then installed the last package of visicut
`
sudo apt install ./visicut_1.9-142-gc05b03f5-1_all.deb`

Then I've run the install of the extension via Visicut

So when I try to use the extension I got the following error message : 
`/usr/bin/python3: can't find '__main__' module in '/home/bocal/.'`
(bocal is the username)

the 4 extension files are present in the exentension folder

- daemonize.py
- visicut_export.inx
- visicut_export.py
- visicut_export_replace.inx

I can read all of them with a text editor.
 noticing that the option "add in visicut" && "open in visicut" extension in inkscape appears twice

mgmax, 2022-10-07:
I tried the same and it works, both with Inkscape 1.1 and 1.2:

- Set up Ubuntu 22.04
- Option A: Ubuntu 1.1 via apt: In the "Ubuntu software" app, remove the snap version. Then select "ubuntu jammy" as source in the top right corner, then install from there  (or you can do the same by 'apt-get install inkscape').
- Option B: Install Inkscape 1.2 via PPA (as you describe above)
- Also install visicut as usual
- Open Inkscape
- draw something
- Extensions - lasercut - open in visicut

Please try to clean your user extension directory. Probably you have an old broken version of the extension in it.

With the snap version of Inkscape, visicut doesn't even install the extension into the right directory, so the extension is not found. But the extension does not work anyway. We should add a warning message in Visicut -- check if `which inkscape` starts with `/snap/` and then inform the user that "snap" is currently not supported.

mgmax, 2024-11-10:
We have the following problems:
- Snap does not allow installing extensions to Inkscape. In theory, it would be possible that we upload our extension to the Inkscape extension manager "store" and the user installs the extension via "Extensions - Manage Extensions". However, the extension manager is currently broken in the snap version: https://gitlab.com/inkscape/extras/extension-manager/-/issues/18
- Snap makes it difficult to call another application from within a snap. (As far as I understand, xdg-open or custom URL handlers should work, see https://forum.snapcraft.io/t/allowing-xdg-open-to-open-files/3789/22 ).
- Ubuntu does no longer provide up-to-date versions of Inkscape as APT package. Maybe because they are taking the APT versions from Debian and there they are also two versions behind. https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1078862

So for now there's nothing we can do except detect the situation and show some instructions for a workaround.

jnweiger, 2024-12-26:
Reproduced with Ubuntu 24.10, inkscape 2.3.1 and VisiCut-2.1-7-g87daf9a5-x86_64.AppImage, (Nov. 24, 2024)

`sudo snap install inkscape` install inkscape using my user home (`/home/jw`) and some system paths /snap/... for settings:

[image]
----

VisiCut installs the inkscape extension under 
```
~/.config/inkscape/extensions/
insgesamt 32
drwxr-xr-x 2 jw jw  4096 Dez 26 18:17 ./
drwxr-xr-x 9 jw jw  4096 Dez 26 18:16 ../
-rw-rw-r-- 1 jw jw   646 Dez 26 18:17 visicut_export.inx
-rw-rw-r-- 1 jw jw 16070 Dez 26 18:17 visicut_export.py
-rw-rw-r-- 1 jw jw   656 Dez 26 18:17 visicut_export_replace.inx
```

jnweiger, 2024-12-26:
Additionally, in my case, VisiCut is installed as appimage.
Calling the binary in the appimage seems another hurdle.

The installed visicut_export.py has `VISICUTDIR = r"/tmp/.mount_visicuXseZBM/usr/lib/visicut"
which was the /tmp/.mount path when visicut installed the plugin. This path differs with a fresh invocation. This results in an error: 

Can not start VisiCut (PermissionError(13, 'Permission denied')). Please start manually or change the VISICUTDIR or VISICUTBIN variables inthe Inkscape extension script

For the scope of this issue, this means, that the visicut_export.py plugin was correctly found.

mgmax, 2024-12-28:
Thanks for another puzzle piece :-)

Do you have the same problem with AppImage also if Inkscape is installed via APT? If yes, please open a separate issue.

For the issue here, I would suggest we first fix the "standard" case where VisiCut is installed normally via DEB and Inkscape is installed via snap. The largest missing piece here is how to call VisiCut from within snap Inkscape. Can you find a solution?
A possible starting point is: You can get a shell inside the snap with `snap run --shell inkscape`. The folder `$SNAP_USER_DATA` is writable from within the snap. Create a `.plf` file (or some other format that is only associated with VisiCut) in $SNAP_USER_DATA and then use `xdg-open` to request opening this file (which is magically redirected to the environment outside of the snap). A dialog appears, asking which program should be used to open the file. Hopefully we find some way so that only VisiCut is suggested in that dialog. A custom URL scheme would be an alternative, though I'm not sure how to transport the data if VisiCut is some day also sandboxed in a snap. https://unix.stackexchange.com/questions/497146/create-a-custom-url-protocol-handler