t-oster/VisiCut #625: How to save laser/material settings?

valkyriesavage, 2021-11-09:
I'm using the most recent Visicut on Mac. My lab doesn't have an existing settings file, so we are trying to develop one that can be shared for our Epilog Helix. Unfortunately, we cannot figure out how to save changes to
> materials (e.g., rendering color of material, thicknesses, etc.)
> power settings (per-material, or any other way)

[image]
Changes I make here are not saved, and I don't see a save button.

[image]
Changes I make here are also not saved, and are reset to 0 power every time.

I don't see instructions in the manual for how to change these things. I tried messing with the .vcsettings files, but they seem to be binary encoded and I can't figure out how to modify them.

valkyriesavage, 2021-11-09:
If it matters, I am building off of the "example" settings.

mgmax, 2021-11-09:
Can you please run Visicut from the command line and copy-paste the output?
I hope the instructions on https://github.com/t-oster/VisiCut/wiki/Development:-Running-from-commandline are still up to date

t-oster, 2021-11-10:
Looks as if you just have to resize the window. The save button is there, but cut off.

valkyriesavage, 2021-11-12:
... sure enough.

It would be nice if the window was automatically sized for that. A student of mine running on his PC has had the same issue with sizing.

valkyriesavage, 2021-11-12:
Hmm... ok, so that successfully lets me change the appearance of the material, but even resizing the window doesn't show any way of saving the laser power/speed settings for cut/mark in my job.

I tried running the commandline instructions, but they do seem to be out of date.

I found VisiCut.MacOS in `/Applications/VisiCut.app/Contents/Resources/Java` , but in spite of updating my version of Java (to 17, after it complained that my version was too out-of-date) I get the following output:
```
[09:11:05][valkyrie@heyrube:/Applications/VisiCut.app/Contents/Resources/Java][]
$ ./VisiCut.MacOS
java version "17.0.1" 2021-10-19 LTS
Java(TM) SE Runtime Environment (build 17.0.1+12-LTS-39)
Java HotSpot(TM) 64-Bit Server VM (build 17.0.1+12-LTS-39, mixed mode, sharing)
Error: Unable to access jarfile /Applications/VisiCut.app/Contents/Resources/Java/Visicut.jar
```

Sure enough, there is no such file or directory on my computer.

t-oster, 2021-11-12:
In your second screenshot you see on the right the edge of a button which should be a "+". If you edit the settings or add some through the + button, there will appear a save button. But those buttons are also not visible in your screenshot because of the window size I guess...

The window layouts seem to be broken on OSX.

Btw VisiCut ships with java 11 and should use only the embedded version. Maybe the instructions are outdated, because we do not seem to have active developers using OSX

valkyriesavage, 2021-11-12:
🙌🏼  Thanks! I found them

closed 2021-11-12 by valkyriesavage