t-oster/VisiCut #626: Generic G-Code Driver does not turn off laser before Z-Axis moves

McNugget6750, 2021-12-03:
Using Visicut 1.9-113-g02234deb.
I transitioned from my own old and modified version (to be found [here](https://github.com/McNugget6750/VisiCut)) to the latest Visicut. My additions were specifically to do better, bidirectional engraving with overscan using the Generic G-Code Driver. Both was added later to the master branch by the core dev group, so I decided to give the new version a shot!

Generally, it works great! Thank you!

### Issue:
I use multiple action layers in my jobs:
- Initiator to heat up the CO2 laser
- Engraving layer
- 3D Engraving layer
- Inner Cut
- Outer Cut

I export this as layers straight from Inkscape. I attached all my settings, templates, an example svg file, as well as the corresponding output.nc file.

If you look for **G0 Z** commands, you will notice that the Z movement comes BEFORE, in the next line, the laser power is set to 0 using **S0**. This leaves my laser on, burning holes into the work.

Example output:
[image]
Please notice in image above that non-Z-Axis G0 movements use S0 just fine but when the Z-Axis is being moved the command is offset by one line.

### Steps to reproduce:
- Load my settings into Visicut
- Import example job into visicut
- For Mapping, select Inkscape-3D
- Check the laser tab and notice, I deliberately messed up the focus settings to illustrate this. Focus on my laser is at 0mm. But to show the issue, I change Z for each layer.
- Export the laser code and open the file using a text editor
- Search for **G0 Z** and notice where the S0 command is being sent. It's in the next line, not coming with the G0 command or even better: BEFORE the G0 Z command line.

[Visicut debugging.zip](https://github.com/t-oster/VisiCut/files/7650321/Visicut.debugging.zip)

**Recommendation: Must fix! Major bug. Safety critical!**

McNugget6750, 2021-12-03:
I should add that "Force laser off during G0 moves" is already on in the settings.
[image]

McNugget6750, 2021-12-03:
Recommended change solves the issue in GenericGcodeDriver.java, replace lines 498 - 504 with:

```
//  protected void setFocus(PrintStream out, double focus) throws IOException {
//    if (currentFocus != focus)
//    {
//      sendLine("G0 Z%f", focus);
//      currentFocus = focus;
//    }
//  }
  
  protected void setFocus(PrintStream out, double focus) throws IOException {
    if (blankLaserDuringRapids)
    {
      currentPower = 0.0;
      if (currentFocus != focus)
      {
        sendLine("G0 Z%f S0", focus);
        currentFocus = focus;
      }
    }
    else
    {
      if (currentFocus != focus)
      {
        sendLine("G0 Z%f", focus);
        currentFocus = focus;
      }
    }
  }
```

t-oster, 2021-12-03:
Hi, thanks for your comment. I will look into it soon. Your proposed patch would not work, because you only set the variable to zero, without outputting anything. However you're on the right track.
I have to see if there are valid use cases for z movement with laser on though, but I don't think it is wanted behavior.   As I understand you the behavior was different in your modified version or is your workflow different now?

McNugget6750, 2021-12-03:
Hi t-oster,

I have everything setup and compiled my code change suggestion and it definitely works!
I set the S0 only when the user wants to force G0 moves with no laser on - which is what I want. I don't see any need to run the laser during G0 - that's what G1 is designed for - at least as far as I'm aware. Please correct me if I'm mistaken!

[image]

As you can see above, this is in line with my expectation of what the code change should be doing. Or am I seeing this wrong?

McNugget6750, 2021-12-03:
Ah, got it. I see your edited comment. I read your comment in the email that came over without the edit. Yes, the above should be fine.

McNugget6750, 2021-12-03:
To answer your second question on workflow:
I just got my Z-Axis table to work so I started testing around with my old Visicut version whether I can move the table to the focus point I want as part of the job. Generally it worked already in the old version. I noticed then that the new version has that baked in, too, and overhauled some parts of the UI (and apparently a lot of stuff under the hood like the overscan behavior I mentioned earlier) so I moved to the new version. Then, I changed my workflow to use different Z-axis focus layers for different job layers. Spoiler: It's absolutely freakin' fantastic!!! This allows so many things and I never had such amazingly perfect engravings and perfect cuts in the same file. So yes, I extended my workflow by adding Z-axis support and then ran into this issue.

mgmax, 2021-12-04:
I agree with the intention of the patch. `blankLaserDuringRapids` should also be respected for Z-moves.

The readability of the code could be improved by restructuring:

(not tested, just a sketch)

```
if (currentFocus != focus) {
   String append = "";
   if (blankLaserDuringRapids) {
      append = " S0";
      currentPower = 0.0;
   }
   sendLine("G0 Z%f" + append, focus);
}
```

McNugget6750, 2021-12-04:
I reorganized my fork and updated to developer and make changes in there that I'd like to create pull requests for.
Unfortunately, the LibLaserCut is a recursive repository and I can't just commit to that after I forked the main Visicut repository.
What's your preferred way of handling this? Should I just fork LibLaserCut, too, update in there, and create pull requests from there?

t-oster, 2021-12-04:
Yes, everything low level like drivers belong to LibLaserCut.

t-oster, 2021-12-04:
You need only to create a pull request against LibLaserCut, I will update visicut to use the latest LibLaserCut after merging.

mgmax, 2021-12-09:
fixed in the next LibLaserCut version: https://github.com/t-oster/LibLaserCut/pull/175

closed 2022-12-24 by mgmax