t-oster/VisiCut #627: Generic G-Code Driver Changes Laser Power in Wrong G-Code Line

McNugget6750, 2021-12-04:
Using Visicut 1.9-113-g02234deb.

Issue:
Raster Engraving jobs are smearing as S-Setting is sent in the wrong line.
This might look like a duplicate of https://github.com/t-oster/VisiCut/issues/626 and it likely is but since the scope is broader, I opened a new ticket

Same template as before, I won't be reposting it as a file:
Initiator to heat up the CO2 laser
Engraving layer
3D Engraving layer
Inner Cut
Outer Cut
I export this as layers straight from Inkscape. I attached all my settings, templates, an example svg file, as well as the corresponding output.nc file. The NC file can be viewed in LaserGRBL and already shows the smearing without the need to engrave it. It's makes the issue pretty obvious.

Example output:
[image]

Steps to reproduce:
Load my settings into Visicut
Import example job into visicut
For Mapping, select Inkscape-3D
Export the laser code and open the file using a text editor or better LaserGRBL.
Either observe the smearing in the gcode viewer or engrave and see it there
[Visicut debugging.zip](https://github.com/t-oster/VisiCut/files/7653302/Visicut.debugging.zip)
.

McNugget6750, 2021-12-04:
For comparison (I might have modified more than I thought when I first started with VisiCut) in my version I have extra values in the configuration that allow me to tell what is Laser 0 and what is Laser OFF. Your config currently only holds What's Laser Max Power.
Then, I did modify the raster code specifically, because back then, it didn't work well since it wasn't overscanning.

Attached are examples that I just exported from my old version of visicut. I also packaged my GenericGCodeDriver.java for reference. Keep in mind, this is a very old code base but it seems it's still relevant as it produces the correct results.

[image]

and here is a comparison between a clean raster output from my version and the output of the current raster output.

Clean:
[image]

Broken:
[image]

[ExportedLaserFiles.zip](https://github.com/t-oster/VisiCut/files/7653387/ExportedLaserFiles.zip)

McNugget6750, 2021-12-04:
My guess is the bug is in LaserCutter.java within
`protected VectorPart convertRasterizableToVectorPart(RasterizableJobPart rp, LaserJob job, boolean bidirectional, boolean useMoveToForWhitePixels, boolean useMoveToForPadding){}`
That's where I implemented the bidirectional rasterizer with overscanning as used in the known good example.

Here is my implementation. I'm looking through the code now...

```
    protected VectorPart convertRasterizableToVectorPart(RasterizableJobPart rp, double resolution, int overscan, boolean bidirectional)
    {
      // Create the result object and preload the first properties and the laser cutter resolution
      VectorPart result = new VectorPart(rp.getLaserProperty(), resolution);
      
      // Calculate the overscan distrance in pixels
      int overscanTemp = (int)Util.mm2px(overscan, resolution) * (rp.getCutDirection() ? 1 : -1);
          
      // Get the first location in the row of pixels in the image
      int y = 0;
      Point lineStart = rp.getStartPosition(y);
      
      // move to the first point of the line minus the overscan compensation
      // and the location of the first non white pixel of this line
      int movetoX = lineStart.x + rp.firstNonWhitePixel(y) + rp.cutCompensation();
      result.moveto(movetoX - overscanTemp, lineStart.y);
      
      // We need to understand what the last X position was during the loops
      // so we store it external to the loops
      int lastX = 0;
      
      // Run through all rows of the image that we want to raster engrave usind gCode
      for (y = 0; y < rp.getRasterHeight(); y++)
      {
        // Check if the line is actually blank and skip it as an optimization step
        if (rp.lineIsBlank(y) == false)
        {
          // Get the first location in the row
          // Question: What is an image and where is it? Is the entire 
          // laser cutter work area considered one raster image?
          lineStart = rp.getStartPosition(y);
          
          /**
           * Here begins the fun part. We want to be (5)mm left of the left most pixel when we start rastering with 0 power.
           * Then we move the the left most pixel and THEN we turn on the power to the desired value.
           * After we have rastered the complete line, we add additional (5)mm of distance with 0 power to let the acceleration roll off.
          */
          
          // Calculate the overscan distance based on the current cutting direction and resolution setting in image pixels
          overscanTemp = (int)Util.mm2px(overscan, resolution) * (rp.getCutDirection() ? 1 : -1);
          
          // Get the first non white pixel in X          
          // Grab its properties so we can modify them
          FloatPowerSpeedFocusProperty overScanProperty = rp.getPowerSpeedFocusPropertyForPixel(rp.firstNonWhitePixel(y), y);
          overScanProperty.setPower(0); // reset the power so that we can actually use overscanning
          // Set the property for the initial line to command
          result.setProperty(overScanProperty);
          // Create a lineTo command for the overscan at the beginning of the line
          movetoX = lineStart.x + lastX + rp.cutCompensation();
          result.lineto(movetoX, lineStart.y);
          // finished running overscan distance
          
          // Now that we have the first non white row as well as the first non white column,
          // we can start rasterizing the line of the image into vector commands
          int x = 0;
          for (x = rp.firstNonWhitePixel(y); !rp.hasFinishedCuttingLine(x, y);)
          {
            // Set the properties for the lineTo command
            overScanProperty = rp.getPowerSpeedFocusPropertyForPixel(x, y);
            result.setProperty(overScanProperty);
            // Calculate the next column (pixel in X) with a different color and set it
            x = rp.nextColorChange(x, y);
            // move to that position with constant speed and the set properties
            result.lineto(lineStart.x + x + rp.cutCompensation(), lineStart.y);
          }
          
          // Store the last X value to use it during the next Y-iteration to use it for the remaining overscan pass
          lastX = x;
          
          // Use the properties of the last engraced pixel and set power to zero for overscann distance
          overScanProperty.setPower(0); // reset the power so that we can actually use overscanning
          // Set the property for the last lineTo command (overscan distance)
          result.setProperty(overScanProperty);
          // Create a lineTo command for the overscan at the end of the line
          result.lineto(lineStart.x + lastX + rp.cutCompensation() + overscanTemp, lineStart.y);
          
          // Switch direction of rasterizing and also switch the pixel compensation as well
          if (bidirectional) rp.toggleRasteringCutDirection();
        }
      }
      return result;
    }
```

McNugget6750, 2021-12-04:
Looking at the same sections in both generation gcode files one can see that I am using the S1 setting I added a lot to power the laser down within the same line of pixels while your code currently keeps the laser on most of the time.

[image]

mgmax, 2021-12-04:
Unfortunately, large code blocks from some old version are not so helpful. I would recommend that you use Git branches in the future so that it is clear what was changed from which version. Or even better, don't keep your changes locally but create a Pull Request so that they are merged back into VisiCut.

I'm not sure where the bug is, since all relevant code parts look innocent:

https://github.com/t-oster/LibLaserCut/blob/cd685394b099d238fdb588c48b4be58865100c3f/src/main/java/de/thomas_oster/liblasercut/drivers/GenericGcodeDriver.java#L875

https://github.com/t-oster/LibLaserCut/blob/cd685394b099d238fdb588c48b4be58865100c3f/src/main/java/de/thomas_oster/liblasercut/LaserCutter.java#L341

https://github.com/t-oster/LibLaserCut/blob/cd685394b099d238fdb588c48b4be58865100c3f/src/main/java/de/thomas_oster/liblasercut/drivers/GenericGcodeDriver.java#L451

https://github.com/t-oster/LibLaserCut/blob/cd685394b099d238fdb588c48b4be58865100c3f/src/main/java/de/thomas_oster/liblasercut/drivers/GenericGcodeDriver.java#L511

Please try to find a minimal example (e.g. only engrave instead of engrave3D, default machine settings, creating a very simple pattern that leads to a short GCode that is obvious to check by hand ...). Then the next step would be to debug the code (breakpoints in the NetBeans debugger, or just add println() commands), to see at which point of the code it goes wrong.

McNugget6750, 2021-12-04:
Hey Max,

Thanks for the very valid feedback. I need to get better at creating pull requests. I have only done with a few times for Marlin Firmware so far. For my old version it didn't make much sense. I'll refrain from posting large code snippets.

I am doing some more investigation and I found something interesting:

- I created a single 3D engraved line in Inkscape and located it at ~10mm. Since all the other engrave and cut modes work, I want to concentrate on analyzing 3D Engrave.
- The behavior of the checkbox "Focus initially on the base plate instead of the material surface" is reversed. With a material thickness of 3 and the checkbox checked, I'd expect the focus to be changed to 3mm for engraving. But it only does that when the checkbox is UNCHECKED.
- Overscan padding is **only** applied if the laser head doesn't cross 0mm. This should be a a user setting. Negative values should be allowed.

**Now the juicy bits:**
- With **Force Laser Off During G0** turned **on**, the laser is **never** turned off for the overscan padding and probably never in general <- Needs more testing.
- With **Force Laser Off During G0** turned **off** the laser is being turned off as expected during G1 moves, however, it has the side effect, that my other issue is back up again https://github.com/t-oster/VisiCut/issues/626. I had a feeling this would just be a workaround for the actual issue. **The laser is not turned off right before Z moves.**
- 
[Single Line 3D Engraving Examples.zip](https://github.com/t-oster/VisiCut/files/7654031/Single.Line.3D.Engraving.Examples.zip)

McNugget6750, 2021-12-04:
Here it's a little easier to see. I marked up the issues:
[image]

McNugget6750, 2021-12-04:
Digging some more, I noticed that Visicus assumes that the laser is usually off during G0 moves and will not enforce.
Reading this, makes me feel like this is a bug in my current grbl version: https://forum.lightburnsoftware.com/t/laser-stays-on-even-for-position-moves/4882/3 as they point out that specifically the laser mode was enhanced since GRBL1.1f. I'm running grbl 1.1d!

According to the linked page:
```
GRBL 1.1f and later have laser mode, set by $32=1, which does two things:
- it adjusts the power output based on the speed of the laser
- it only fires during G1 moves, not G0 moves
```

Laser mode is already active on my machine but the change on the machine side does make sense and would solve all my issues.

I realized this must be the case when I created another example with two engraved lines and one 3D engraved line. The laser, again, would not turn off after the first engraved line while moving to the second one. No engraving would ever work properly then. But that didn't make sense either. I'll investigate more but I'll also try to upgrade my machine today to see if this solves the issue.

[image]

mgmax, 2021-12-04:
Looks like you're getting close to the source of bug and just need a little more debugging to find out why. My best guess would be that this line is to blame for some of the missing S-values:

https://github.com/t-oster/LibLaserCut/blob/cd685394b099d238fdb588c48b4be58865100c3f/src/main/java/de/thomas_oster/liblasercut/drivers/GenericGcodeDriver.java#L502

Assume blankLaserDuringRapids == true. Then, move() sets currentPower=0. A subsequent line() with power=0 will then not output an S-value, which is incorrect.

The correct version should be "currentPower = -1" (invalid value), so that the next line() will force that S is sent again. We should possibly make this change even for `blankLaserDuringRapids == false`, because it does not hurt if S is sent slightly too often.

Just as a note: If "force laser off during G0" is disabled, then VisiCut assumes that G0 temporarily switches off the laser until the next G1, which will revert to the previous S value.

I don't have enough time to try all this, so it would be great if you could have a look and submit it as a pull request.

McNugget6750, 2021-12-04:
Thanks for the detailed response! I'll look further into it as I also believe there should be more S-commands being sent rather than less. Laser safety is the most important thing when working with lasers and having lasers on during times where they should not be on, should be avoided at all cost.
Having said that, GRBL1.1h does solve this issue! It interprets the commands properly and deactivates the laser during all G0 moves no matter what, once laser mode is activated using $32=1

I have one more problem to solve though but I'll open another ticket for that: The checkbox for "Focus is initially on the base plate instead of the material surface" has no effect in the current build. The Z-axis always moves to 3mm at the beginning of the job for a 3mm material no matter that's checked or not.

McNugget6750, 2021-12-04:
I implemented the change you proposed but the effect has not shown, yet:
[image]

mgmax, 2022-09-18:
should have been fixed in https://github.com/t-oster/LibLaserCut/pull/175

closed 2022-09-18 by mgmax