t-oster/VisiCut #628: 3D Engraving Overscan in Negative Coordinates Needed

McNugget6750, 2021-12-05:
I'm using 3D engraving and noticed that overscanning does not allow moving into negative coordinate space.
This causes issues when engraving close to X0.
In many cases this is not an issue, I understand. But almost all machines use arbitrary coordinates that are not necessarily related to the physical limitations of the motion platform anyways.
It should be an option to allow negative coordinate space padding / overscanning.

I can implement it but I'd like to as for a pointer to the right location. I assume, I have to go back to
`convertRasterizableToVectorPart()` and see how the min and max values are implemented. It wasn't obvious to me, yet.

McNugget6750, 2021-12-05:
I think I found it: 
LaserCutter.java 359: `        preStartX = Math.min(rightLimitPx, Math.max(leftLimitPx, preStartX));`
LaserCutter.java 386: `        postEndX = Math.min(rightLimitPx, Math.max(leftLimitPx, postEndX));`

I'll play with that a little bit.

mgmax, 2021-12-05:
Currently, this is on purpose to allow for engraving close to the left edge of the laser bed without hitting some mechanical endstop. VisiCut assumes that the laser bed size is the limit of allowed coordinates. For relative positioning you can use right-click -> set start position. It is supported at least on Epilog Laser devices, but I'm not sure if anyone has ever used that feature in real life.

We'd need to add a parameter for the allowed overrun distance to the laser device and consider that in `convertRasterizablePartToVectorPart`. Feel free to just modify the constant inside the function for first experiments. Even then I'm not sure if some other checks in LibLaserCut will complain if a coordinate is outside of the laser bed.

McNugget6750, 2021-12-05:
If nothing else complains, I should be able to simply adjust the lines above and get into negative space. In thought the hardware limits might be the reason for this. My laser homes and moves a few mm away from home on purpose to allow for this kind of stuff. I also move the head around for different applications and redefine XY=0 there. In that specific case, having no access to negative coordinates is a completely arbitrary limitation.

Once I have it working, I'll add a configuration setting to the GenericGCodeDriver.java so that the allowable negative padding can be specified by the user.

mgmax, 2022-12-20:
fixed by https://github.com/t-oster/LibLaserCut/commit/6b6c86d93925018f1d79432e06c5b0ff799a869f

closed 2022-12-20 by mgmax