t-oster/VisiCut #629: New builds don't seem to work with LAOS Lasercutters

thinkl33t, 2021-12-15:
I have access to two LAOS lasercutters, one at Hackspace Manchester, and one at home.   Both running the latest version of laos.

We've noticed that on new versions of visicut (those from the visicut downloads page), what looks like completely blank files are being sent through to the laser cutter.  Symptoms are that when asked to run, the laser will start up the fans, extraction etc, then immediately end the job.  

I have inspected the simplecode files visicut is generating (by exporting them) and they seem to be correct.  Some files seem to be generated and sent through correctly - notably the finger-box example works fine.

We have triaged all the visicut versions we could get our hands on, on debian and macos, and found the following:

* 1.8-103 works fine (we had this version lying around)
* 1.9-61 works fine (same)
* 1.9-70 works fine (from the github releases page)
* 1.9-113 doesn't work
* 1.9-111 doesn't work

mgmax, 2021-12-15:
Is this the issue? https://github.com/t-oster/LibLaserCut/issues/131
Then the expected symptom would be that it does not work for all files.

You could try with network sniffing software (e.g., wireshark) to check what's sent to the cutter and check if the length roughly matches the simplecode file.

thinkl33t, 2021-12-15:
[image]

[image]

thinkl33t, 2021-12-15:
Attached is a zip of the pcapng file if you want to look - this is using visicut 1.9-135

[visicut-1.9-135.zip](https://github.com/t-oster/VisiCut/files/7720687/visicut-1.9-135.zip)

thinkl33t, 2021-12-15:
Same file, 1.9-41 - worked fine

[image]

[visicut-1.9-41.zip](https://github.com/t-oster/VisiCut/files/7720811/visicut-1.9-41.zip)

t-oster, 2021-12-15:
Can you set up a dev environment and do a git bisect in LibLaserCut in order to find the commit, which introduced that problem?

mgmax, 2021-12-15:
Does it make a difference if you restart VisiCut before every laser job? We sometimes had bugs regarding the initialization, where the laser driver only worked correctly for the first job.

thinkl33t, 2021-12-15:
Just tried -135 again with wireshark doing its thing, same behaviour on the first attempt as subsequent attempts.

mgmax, 2022-09-03:
should be fixed in the next release, same issue as https://github.com/t-oster/VisiCut/issues/647 - files only work if they are big enough (and/or if you have random luck with the Java garbage collector timing)

closed 2022-09-03 by mgmax