t-oster/VisiCut #631: Request for documentation: how to export SVGs from Adobe Creative Cloud to work with visicut

valkyriesavage, 2022-01-06:
I recently had to upgrade my Illustrator from CS6 to Creative Cloud, but now I can't seem to generate any SVGs that work with Visicut. I've uploaded a screenshot of the dialog that Illustrator offers for this export, as well as the dropdown for CSS style (which I do get a warning is not supported in Visicut).

[image]
[image]

The best I can do is use "Style Elements" for the SVG, which renders a black box the size of my drawing
[image]

All the other options for CSS seem to result in a 0-size drawing with no properties.

I've also attached the SVG and Illustrator file in question.
[Archive.zip](https://github.com/t-oster/VisiCut/files/7820901/Archive.zip)

I would like to know how to configure this dialog to get SVGs that work in Visicut.

The "Open With Visicut" script for Illustrator does still seem to work, so this isn't urgent. It would still be helpful. :)

mgmax, 2022-01-06:
It's basically the same issue as with CorelDraw: VisiCut does not support CSS classes, only style attributes.
https://github.com/t-oster/VisiCut/issues/339
Here's the closest point in the source code:
https://github.com/t-oster/VisiCut/blob/9e819d34b501d015cc9fe7bc214ca3516db410b7/src/main/java/de/thomas_oster/visicut/model/graphicelements/svgsupport/SVGObject.java
The issue is with the underlying SVGSalamander library, which does not support CSS.
https://github.com/blackears/svgSalamander/issues/75

"Style attributes" or possibly "Presentation attributes" should work. Could you please upload SVGs created using these variants?

You can also have a look at the script file and see what it does.
https://github.com/t-oster/VisiCut/raw/master/tools/illustrator_script/OpenWithVisiCut.scpt
I can't open it because I don't have Illustrator.

valkyriesavage, 2022-01-06:
Iiiiiiiiinteresting, the issue seems to be something else; when I just now reopened visicut with SVGs saved in that manner (attached), they worked fine.
[Archive 2.zip](https://github.com/t-oster/VisiCut/files/7821189/Archive.2.zip)

I can't figure out what the error was? (I tried reopening the bugged svg first, then opening those, but it seemed to work fine doing that, as well.)

Ok, at any rate, this works for me. Thanks!

closed 2022-01-06 by valkyriesavage

mgmax, 2022-01-06:
I added the info to use "Style attributes" to the manual. Would be nice if you could have a look - feel free to edit and improve the manual, it definitely needs work.
https://github.com/t-oster/VisiCut/wiki/VisiCut-manual