t-oster/VisiCut #632: Engraving area

LaFabrique35, 2022-01-14:
Hello, I would like to optimize laser engraving for my users using VisiCut.
My laser travels in raster over the entire engraving area, including areas where there is nothing to engrave. Do you have a solution to optimize this and save burning time? My laser cutter works with a SmootieBoard driven by OctoPrint
Thanks !

LaFabrique35, 2022-03-10:
Does anyone have a solution for me?

mgmax, 2022-03-19:
What you are describing may be normal. For every horizontal line, the laser starts some centimeters left of the leftmost engraved ("black") pixel and ends some centimeters right of the rightmost engraved pixel. 

Everything else would be super complicated to program and only be supported by half of the lasercutters that VisiCut supports. Splitting into multiple areas would often not be faster because the acceleration and deceleration phase ("overscan" left/right of the actual engraving) takes significant time. 

Sometimes it helps to rotate your design or move the engravings so that they are on one line.

Some further possible workarounds, which are however slightly complicated:

You could split your design into pieces that you add separately (using "add" instead of "open") and send together. Then they should be sent separately. 

Or have several engraving layers that each map to a separate engraving profile (e.g., "engrave 2").

LaFabrique35, 2022-03-21:
Ok, thank you for these answers.

closed 2022-03-31 by mgmax