t-oster/VisiCut #633: Fix position file X=0 Y=0

LaFabrique35, 2022-01-14:
Hello, when I import an SVG file into VisiCut, I would like it to automatically position itself at X = 0 and Y = 0 on the work surface. The settings are correctly configured in the options but the file is not positioned at the correct coordinates.

t-oster, 2022-01-14:
How is your SVG view box? Usually the view box is positioned 0/0 so you can absolutely position elements in the svg

LaFabrique35, 2022-01-14:
My SVG view box is positioned with an origin at the bottom left. Ditto on Inkscape. How to force the file to be positioned in X 0 Y 0 during the import?

t-oster, 2022-01-14:
So it is top left an you're expecting bottom left? We should make this corresponding to the Lasercutter settings. Until then just make the SVG view box fill the laser bed