t-oster/VisiCut #634: Downloading VisiCut with Edge on Win11

mgmax, 2022-01-15:
I noticed that VisiCut can't be downloaded on Win11 with Edge because it gets blocked. Downloaded it successfully with Firefox. 

Probably there is a workaround.
Maybe something we should document for users?

_Originally found by @sebastian-schmid in https://github.com/t-oster/VisiCut/issues/624#issuecomment-982734580_

t-oster, 2022-01-19:
I just downloaded it with a fresh Win-11 dev VM and Edge. You have to click **keep** and **I know what I am doing** a few times, but it works. The reason they say is that it is not downloaded very often:
[image]
[image]
[image]
[image]

mgmax, 2022-01-19:
Code signing would probably help, but as far as I know one has to pay for a certificate and this can't be automated.