t-oster/VisiCut #638: Wainlux K6....  IOException: Port in use: COM3

schmiernippel, 2022-02-06:
On my Wainlux K6 Laser i get this Error:

 IOException: Port in use: COM3

A lot of people alsom get this error....

Any help or workaround??

greets

t-oster, 2022-02-06:
Can you screenshot your device manager to confirm the Lasercutter is present on COM 3?

t-oster, 2022-02-06:
You could also try to run visicut as admin

schmiernippel, 2022-02-06:
[image]

I have also tried run as admin and different baudrates.....

thanks

t-oster, 2022-02-06:
Is there any other program opened which uses the com Port?

schmiernippel, 2022-02-06:
No....and i also have tried an other PC....and a get the same error...

schmiernippel, 2022-02-06:
This Error i also get:

“IOException: Underlying input stream returned zero bytes”.

schmiernippel, 2022-02-06:
> Is there any other program opened which uses the com Port?

I will try changing the port....

schmiernippel, 2022-02-06:
CHanging the port will not help...

t-oster, 2022-02-06:
Does it work with any other Software?

schmiernippel, 2022-02-06:
Yes...it works with original software....and also with differen versions of the software!

schmiernippel, 2022-02-06:
Is there anything i can do for you that you know what is not working??

t-oster, 2022-02-06:
Hmm. Okay. Are you able to set up a development environment so we can debug step by step?https://github.com/t-oster/VisiCut/wiki/Development:-Getting-started

mgmax, 2022-02-06:
Can you connect to the serial port using other software (HyperTerminal, Putty or similar)?

Can you find out which program blocks the serial port? https://www.equisys.com/Support/technotes/howto-identifying-which-process-is-using-the-serial-port

schmiernippel, 2022-02-06:
> Hmm. Okay. Are you able to set up a development environment so we can debug step by step?https://github.com/t-oster/VisiCut/wiki/Development:-Getting-started

Sorry...no...thats over my knowledge :-)

schmiernippel, 2022-02-06:
> Can you connect to the serial port using other software (HyperTerminal, Putty or similar)?
> 
> Can you find out which program blocks the serial port? https://www.equisys.com/Support/technotes/howto-identifying-which-process-is-using-the-serial-port

yes i can access the port with other software, checked with "serial port monitor from eltima software"
it works also with original laser software

When i open visicut and send the code to the  engraver, a can not connect with other software to the Port, so visicut it is connected to the Port.

robin-toolmaker, 2022-02-10:
Same “**IOException: Underlying input stream returned zero bytes**” here.
VisiCut --debug sagt:
------------
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.dom4j.io.SAXContentHandler (file:/C:/Program%20Files%20(x86)/VisiCut/Visicut.jar) to method com.sun.org.apache.xerces.internal.parsers.AbstractSAXParser$LocatorProxy.getEncoding()
WARNING: Please consider reporting this to the maintainers of org.dom4j.io.SAXContentHandler
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
using single-instance port: 6546
java.io.IOException: Underlying input stream returned zero bytes
        at java.base/sun.nio.cs.StreamDecoder.readBytes(Unknown Source)
        at java.base/sun.nio.cs.StreamDecoder.implRead(Unknown Source)
        at java.base/sun.nio.cs.StreamDecoder.read(Unknown Source)
        at java.base/java.io.InputStreamReader.read(Unknown Source)
        at java.base/java.io.Reader.read(Unknown Source)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.waitForACK(K3EngraverDriver.java:294)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.sendLine(K3EngraverDriver.java:310)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.send4ByteCommand(K3EngraverDriver.java:319)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.sendConnectSequence(K3EngraverDriver.java:407)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.sendJob(K3EngraverDriver.java:551)
        at de.thomas_oster.visicut.VisicutModel.sendJob(VisicutModel.java:778)
        at de.thomas_oster.visicut.gui.MainView.lambda$executeOrSaveJob$7(MainView.java:2209)
        at java.base/java.lang.Thread.run(Unknown Source)
java.io.IOException: Underlying input stream returned zero bytes
        at java.base/sun.nio.cs.StreamDecoder.readBytes(Unknown Source)
        at java.base/sun.nio.cs.StreamDecoder.implRead(Unknown Source)
        at java.base/sun.nio.cs.StreamDecoder.read(Unknown Source)
        at java.base/java.io.InputStreamReader.read(Unknown Source)
        at java.base/java.io.Reader.read(Unknown Source)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.waitForACK(K3EngraverDriver.java:294)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.sendLine(K3EngraverDriver.java:310)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.send4ByteCommand(K3EngraverDriver.java:319)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.sendConnectSequence(K3EngraverDriver.java:407)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.sendJob(K3EngraverDriver.java:551)
        at de.thomas_oster.visicut.VisicutModel.sendJob(VisicutModel.java:778)
        at de.thomas_oster.visicut.gui.MainView.lambda$executeOrSaveJob$7(MainView.java:2209)
        at java.base/java.lang.Thread.run(Unknown Source)

JarrettR, 2022-08-11:
Also getting this. Windows and K6.

Is there anything we can do to help debug? I do have a logic analyzer, if necessary, although I don't particularly want to pop the top off the unit if I don't have to.

mgmax, 2022-08-12:
I'm somewhat confused by the comments - the original poster says "IOException port in use", another one says "Underlying input stream returned zero bytes". Please only discuss the "Port in use" issue here.

For the "zero bytes issue", see https://github.com/t-oster/LibLaserCut/issues/170 .

stas00, 2024-12-13:
On linux `chmod a+rw /dev/ttyUSB0` overcomes this problem when running VisiCut as a normal user.

Change `/dev/ttyUSB0` to whatever your Linux assigns when you connect the usb cable to the engraver.

To get the right tty file something like this should work:
```
dmesg | awk '/tty/ && /USB/ {print "/dev/"$NF}'
```

see more here: https://unix.stackexchange.com/questions/144029/command-to-determine-ports-of-a-device-like-dev-ttyusb0

mgmax, 2024-12-21:
> On linux `sudo chmod a+rw /dev/ttyUSB0` overcomes the "IOException port in use" problem when running VisiCut as a normal user (at least till the "Underlying input stream returned zero bytes" occurs and then it comes back).

Good find! Let's break down the problem:

1. The basic issue is that your user has now write access to the serial port.
The correct solution is to add your user to the `dialout` group
```
usermod -a -G dialout $USER
```
and then log out and back in again (or reboot if that doesn't help).

2.  Looking at the [code](https://github.com/nyholku/purejavacomm/blob/master/src/purejavacomm/PureJavaSerialPort.java#L1081-L1090), it seems that the purejavacomm library gives a misleading `PortInUseException` when actually it has no write access to the device.
3. LibLaserCut / VisiCut simply shows the message "port in use", not warning about further issues.

stas00, 2024-12-22:
Indeed, the correct solution here is for VC to check that the device path it intends to use has the correct permissions before attempting to use it - it has the path from its config (manual or automatic search). And inform the user how to overcome the problem if the perms are wrong.

theohm, 2025-02-15:
The two exceptions “Port in use: [...]” and “[...] zero bytes“ Seem to be linked up. On Linux, when I hit the Execute button, I get the “Underlying input stream returned zero bytes“ exception. When I ht the Execute button again, I get “Port in use: /dev/ttyUSB0“.

BerthB, 2026-08-25:
Hi There. Any progress or info about implementing the special protocol? I have the exact same problem.