t-oster/VisiCut #640: Baseplate moving down unexpected when starting my job

bendreality, 2022-02-11:
Hey Guys, 

yesterday I ran into a strange problem. I wanted to cut 4mm ply. I leveled the platform so the material surface is in focus. But, when I start the job the baseplate is moving down

My Settings in german :

visicut version: 1.9-138.g2801437a
Materialstärke (Material thickness): plywood 4mm
Fokus ist anfangs auf der Grundplatte... (Focus on Bed level): not checked
Einstellungen für cut > focus (focus offset in Laser Settings): 0

I would expect that the Print Plattform not moving at all when I start the job with this settings, but it's moving down the material thickness of 4mm and starts cutting out of focus as if I checked the "Fokus ist anfangs auf der Grundplatte... " option. 

With the laptop of a friend running an older version of visicut everything worked as expected.  

Am I doing somethnig wrong or is it a bug?

greeting form Leverkusen
Ben

t-oster, 2022-02-14:
Hi, can you tell the exact "older version" where it's working? We can then check the differences in the driver

t-oster, 2022-02-14:
also please export your settings and load them up somewhere, so we can start with exactly your settings

bendreality, 2022-02-17:
Hi,

today I'm at the workshop with the others and check out what the "older version" exactly is. I'll send everything tomorrow. 

thx
Ben

bendreality, 2022-02-20:
The working Version was the 1.8. But I can't tell the exact build because It was also updated last week to the current 1.9. 

Here ist a link to my settings and 2 Screenshots:
[Download Settings](https://drive.google.com/drive/folders/11ujM20XWXZR04xEo0ZwXggHtz6lRGpv9?usp=sharing)

thx
Ben

bendreality, 2022-02-25:
Anything new on the topic?

t-oster, 2022-02-25:
In my first test it looks as if the checkbox does not do anything at all, however in my test the baseplate was never moving instead of always... I suspect some incompatibility with the old Beans binding framework, but I need some more time to investigate

t-oster, 2022-02-25:
Hi @bendreality, could you test the 1.9-141 version?

bendreality, 2022-02-28:
Sure thing. Will do. I'm at the workshop next Thursday and do a test then.

t-oster, 2022-03-08:
Did it work?

bendreality, 2022-03-09:
Damn, totally forgot to answer. Sorry for that. 
Everything worked as expected. Thanks for your support.

t-oster, 2022-03-09:
ok, so this can be closed then.

commit e75194f3 (2022-02-25): Bug: Missing getter for useThicknessAsFocusOffset causes the checkbox to not sync with the model.
Should fix #640
closed 2022-03-09 by t-oster