t-oster/VisiCut #644: Visicut generate integer value for power in G-code export

danke-0x21, 2022-07-19:
Hi !

I upgraded Visicut from **1.9-102** to **1.9-142** version then I saw in all my new generated gcode files that the power setting is only set to 100% (if above 50%) or 0% (if below 50%).

For instance, a G1 command at 60% should be like `G1 X1 S0.6` but turns into `G1 X1 S1` instead.

Here are gcode of a 20x20mm square from **1.9-102** version with power set to 50% and speed to 100%...

```
G92 X0 Y0 Z0  
G0 X0.304800 Y20.352400 F6000  
G1 X20.320000 Y20.352400 S0.500000 F6000  
G1 X20.320000 Y0.337200  
G1 X0.304800 Y0.337200  
G1 X0.304800 Y20.352400  
G1 X0.304800 Y20.352400  
G0 X0 Y0
```

...and same from the latest **1.9-142** version

```
G92 X0 Y0 Z0
G0 X0.3048 Y20.3524 F6000
G1 X20.32 Y20.3524 S0 F6000
G1 X20.32 Y0.3372
G1 X0.3048 Y0.3372
G1 X0.3048 Y20.3524
G1 X0.3048 Y20.3524
G0 X0 Y0
```

So I downgraded Visicut from 1.9-142 to 1.9-141, 1.9-139, 1.9-138 with same trouble.
The I downgraded to 1.9-102 version and everything work properly again.

I checked all available parameters in Visicut to identify any config problem with no result.

Tested on Windows with 2 disctinct PCs.

t-oster, 2022-07-19:
Hi, please provide the exact steps to reproduce:
- Which settings to import (or a zip file if not publically available)
- Test file to load
- which driver are you using
- etc...

t-oster, 2022-07-19:
I already can see: It was commit 360f95055ec7db490d5598a7eee3dac65951efb8 which introduced a property "sCodeDigits" which defaults to 0 (why?). You should be able to set those for your driver to 4 or higher to get the correct g-code

t-oster, 2022-07-19:
https://github.com/t-oster/LibLaserCut/commit/360f95055ec7db490d5598a7eee3dac65951efb8

t-oster, 2022-07-19:
Should be fixed in the next version. However if you saved your settings with a version in between you may have to adjust the settings manually.

closed 2022-07-19 by t-oster

danke-0x21, 2022-07-19:
Thanks for fast fix even if I didn't undertand what you have done.
In my case I run visicut for smoothieboard based lasercutter, does it change anything to your fix ?

>  you may have to adjust the settings manually.
What do you mean ? Where can I change manually this setting if I want to keep using the latest 1.9-142 version?

t-oster, 2022-07-19:
Just try the latest version 1.9-143 (should be out in 5 minutes).

t-oster, 2022-07-19:
ok sorry... there is still a problem with the tests. I won't have time to fix it this week.... 
Just go to the lasercutter settings and change the value in the screenshot:
[image]

t-oster, 2022-07-19:
[image]

danke-0x21, 2022-07-19:
I took the power back ! ;)
Thank you a lot !

```
G92 X0 Y0 Z0
G0 X0.3048 Y20.3524 F6000
G1 X20.32 Y20.3524 S0.5 F6000
G1 X20.32 Y0.3372
G1 X0.3048 Y0.3372
G1 X0.3048 Y20.3524
G1 X0.3048 Y20.3524
G0 X0 Y0
```