t-oster/VisiCut #645: Epilog Laser zing pinging but no connection from visicut

hirilpatel, 2022-07-20:
Hello,

I recently had problem with my laser cutter for ip address network issue which is connected via lan to my wifi router. So in order to avoid them I have assigned a reserve IP address in my router for the laser cutter. Now the problem is that I am able to ping the laser cutter but when sending the job via visicut it says connection time out. can someone advice me what to do with this issue?

mgmax, 2022-07-20:
That's highly unusual and it's 99% certain that VisiCut is not at fault. Please check:
- Is the IP configured correctly? Note that the Epilog Zing does not DHCP (automatic IP config), you must set it manually.
- Does pinging fail if you unplug the lasercutter? If it still works, you are pinging something else.
- Does your router have some kind of filtering that denies the communication? (e.g., AVM FritzBox has some restrictive settings for the guest WiFi like "allow Wifi guests to communicate with each other", "allow all applications to WiFi guests")

hirilpatel, 2022-07-20:
1, Yes IP is configured correctly,(there would be no pinging i guess if not)(the connection that  i am pinging is laptop>wifi>lan>epilog. Yes I think EPILOg does not support DHCP hence I maually assigned(reserved) IP address in my router for epilog zing(using its MAC address)
2. Yes it fails.
3. i have my router configured for most basic setting(quick setup) so no filtering was set in order to able let use guest the laser cutter without hassel.

mgmax, 2022-07-20:
Then please try
- over LAN via your router
- over LAN with a direct connection between laptop and lasercutter (needs a manually configured IP for your laptop)

to completely exclude any network issues.

Also, note that you probably can't use USB and Ethernet together on the Epilog Zing, you may need to disconnect USB and restart the laser before Ethernet works again.

hirilpatel, 2022-07-25:
Ok so Here is what i  have tried now,

I did ping test via lan(diretly from computer to laser) and (epilog-lan-wifirouter-wificonnection-computer) both were successful. 

Now I tried to laser cut by using direct lan connection with computer but no success, visicut says connection timeout.

Same with using wifi router

now I tried to print test page from windows printer setting( please see attached image for what i mean) and its engraved test page as expeted on a wooden sheet(i don't remeber the nameof the material).

So can you suggest me what else should i do?
[image]

mgmax, 2022-07-25:
That's weird. You will have to look at what's happening using Wireshark or similar network capturing software. Maybe Visicut uses the wrong host/port?

hirilpatel, 2022-07-25:
how do you see host(port) not the IP address of printer? I know how to see host on visicut so I would like to know on laser cutter side. And what specific should I be seeing on wireshark(is it free)?

hirilpatel, 2022-07-26:
Well now it works, the problem was from host name, apparently i though host name is different that ip address, but after changing to corrct ip address in host name field it works. thank you for your support

closed 2022-09-03 by mgmax