t-oster/VisiCut #647: Camera Calibration sends empty Job File to Laser Cutter.

mariolukas, 2022-09-02:
VisiCut Version : 1.9-141
Operating System MacOS: Monterey 12.2 
System: MacBook Pro (M1)
LaserCutter: LAOS based.

I am trying to calibrate my camera.  Installed VisCam successfully. I am getting a camera image etc. 

My issue: 

- clicking the send button on the camera calibration dialog. 
- job was send is confirmed
- job arrived at the laser cutter and is showing up on the display
- confirming the job with the laser cutter controls ( OK Button)
- laser cutter is homing
- nothing else happens

Regular cut and mark jobs are working great. I have checked the job data on the SD card. It seems that visicut is sending empty jobs.

mgmax, 2022-09-03:
I found the issue, working on it

mgmax, 2022-09-03:
@t-oster can you please trigger a build? Travis CI still seems broken

closed 2022-09-03 by mgmax

mgmax, 2022-09-03:
The new version is online, please give feedback whether 1.9-146 works

mariolukas, 2022-09-04:
Thank you, it seems to work now.