t-oster/VisiCut #661: Change application name and titlebar name

mu-arch, 2022-12-24:
I'm deploying Visicut in some schools and I need to change the name to something more generic like "Laser Cutter Tool". I have no experience with Java and have struggled to achieve this.

I tried:
- Editing the name tag inside pom.xml
- Looking through VisicutApp.java in visicut.gui

I assume it pulls the name out of a class somewhere and that's why I'm not seeing it. Any help appreciated.

mgmax, 2022-12-24:
Could you please explain in more detail why you need this change and what exactly you want to change (e.g., only the desktop shortcut)? I have personally held many workshops with school children and there was never a problem about the naming of VisiCut. Children can learn and adapt very fast, and many other apps also have specific names (e.g. Excel and not "Spreadsheet Calculation Tool").

mgmax, 2022-12-24:
Followup comment: From the issue title I see that you want to change "Application name and titlebar name". I assume this means that a normal user would no longer see the name "VisiCut". Personally I find this unethical because this hides the work of the original authors. People should be able to find the software and also use it on another computer. Therefore I am not willing to help with such changes.

mu-arch, 2022-12-25:
Understandable. It's not so much hiding the creator because the about menu is intact with all license and credits. It's more because we want students to just click the icon that says "Laser Cutter".

I did manage to find where to change the name, however when a file is opened it reverts the name to Visicut - filename.ext so that must be getting set somewhere else.

Thanks for the replies, I will close this issue.

closed 2022-12-25 by mu-arch