t-oster/VisiCut #662: PDF Support

mu-arch, 2022-12-24:
I've been trying to figure out if it would be possible for me to implement this feature. Some Chinese cutters I've used have been able to take a non vectorized PDF file and figure out how to print from it, so it must be possible. I don't have any experience with Java so any suggestions for libraries to use would be appreciated.

mgmax, 2022-12-24:
You may want to have a look at EPSImporter, which converts from EPS to SVG using https://java.freehep.org/ for the import. There is a related PDF library that you could use with (hopefully) little code changes.

If you are looking for a workaround?
Which software are you using; maybe this software can also save to SVG?
As a workaround, open the PDF with Inkscape and use the VisiCut Inkscape extension.
https://github.com/t-oster/VisiCut/wiki/VisiCut-manual#application-settings-for-exporting-to-visicut
https://github.com/t-oster/VisiCut/wiki/VisiCut-manual#inkscape-and-illustrator-extensions

mu-arch, 2022-12-25:
Actually let me step back and address my bigger overall issue. I want to support the ability for people to "Print to Laser Cutter" in a Windows program like illustrator. I've looked for a way to create a virtual printer in Windows, and while I had some leads it's like Chinese to me mostly. Therefore I settled on using a program called "ClawPDF" which will save either a PDF or PNG to a folder which a Rust program I have written picks up and opens in Visicut.

Would it be possible for Visicut to support this natively? Sadly I have no experience with Java so I can't really do anything myself.

mgmax, 2022-12-25:
To be honest, writing a windows printer driver sounds complicated and not fun. Trying to vectorize the input even more.

For Inkscape, we have an extension that directly sends to VisiCut. You can even configure a shortcut for that in the Inkscape settings.

For Illustrator, we also have such an extension in the past. As far as I know it no longer works with the newest version, and also the "Install illustrator extension" entry in VisiCut fails / is disabled because the new installation path is not found. Feel free to improve it and tell us the correct installation folder for newer versions. https://github.com/t-oster/VisiCut/tree/master/tools/illustrator_script

mgmax, 2022-12-25:
As an alternative, you could just save the SVG from Illustrator and open that with VisiCut. If children can learn how to use a lasercutter, then they can also learn how to do that if you just print out the screenshots of what to do.

mu-arch, 2022-12-25:
Ahhh if only life were that simple, the bureaucracy at our school makes these requirements, not me. Alas I will trudge on and try to get something shimmed together. Perhaps my connector program could be added to the repo if others might find it useful.

mgmax, 2022-12-27:
Okay, so I would close the issue for now. Feel free to respond and maybe reopen, as soon as there is something new. Let us know what you did.

This vectorization tool could come especially handy for "non-digital" input: I know of other FabLabs who had workshops where people just painted with big pens onto paper. All analog input :-) . This was then scanned, vectorized and sent to VisiCut, to cut the desired shape from wood or acrylic. Here, if the scanner places the PDF in a folder, the tool could automatically find the PDF and put it into VisiCut.

Something similar can be done with silhouette photos.  https://github.com/fau-fablab/zeichnen-bauen  https://github.com/fau-fablab/zeichnen-bauen-windows

closed 2022-12-27 by mgmax